prusa3d/Prusa-Firmware-Buddy #13: System crash after long idle period

WarrenSchultz, 2019-12-22:
Firmware 4.0.1.
Printed a benchy, plugged into the network, realized that the network doesn’t do anything, left idle. After an unknown number of hours, the system crashed.
[image]

schiffer-soft, 2020-02-07:
Hi

Self problem. But not afer long idle period.
While printing I want to view the web connect site.
First connect worke fine. After 2 minutes I refresh the site.
Then print stopped with bluescreen.

[image]

MFG Chris

JohnnyDeer, 2020-06-09:
Hi, bluescreen bug is already fixed on newest versions of firmware, i´m closing this issue.

closed 2020-06-09 by JohnnyDeer