prusa3d/Prusa-Firmware-Buddy #15: Load Filament and Color Change Bug

akariin, 2019-12-22:
Load Filament - correct color extruding question only prompts once. If you select “no”, more filament is extruded, but the question doesn’t repeat to confirm. The load filament routine completes, thus requiring to start the load filament process if something was wrong. In pre printing preparation stage, this is ok. But, during mid print, if the user/printer fails to load the filament, then the printer resumes printing once the load filament routine is complete and can print without filament for a few layers until it reaches the extruder.

While trying to find an option to restart the load filament process, I found the Change Color Filament option in the print options, but the option did not work. Eventually the printer crashed with E1 error, with nozzle and bed no longer heating prior to the error. Restarted printer and removed fail print, printer is still ok.

Load filament initial load, if the user/loading gear fails to grab the filament initially, the pre-set load movement from the gear to extruder can be missed, then the user would have to restart the printer to redo it.

tkris-sd, 2019-12-22:
I've also experienced the first part of this issue - "Is color correct" prompt needs to be repeated until color is actually correct. Filament was 20-30mm away from hot end when printing resumed. I aborted the print as it would have been 3+ layers on the small print before the filament was at the hot end.

ryanpriore, 2019-12-24:
I have also experience the first part of the issue: "Is color correct" prompt needs to be repeated until color is actually correct.

michalxfanta, 2020-01-07:
We are working on adjusting the filament loading behavior. The fix will appear in one of the other releases.

closed 2020-01-07 by michalxfanta