prusa3d/Prusa-Firmware-Buddy #16: Y-Homing Crash

Kasperc88, 2019-12-22:
Sometimes when homing with G28 the printer can't detect the end of the y-axis and therefore the y-motor will keep crashing against the end for 2 - 3 seconds before it finds out its the end.

Turning the power off and on helps. But the issue keep coming back.

michalxfanta, 2020-01-07:
Which firmware version (with build number) you use?

Kasperc88, 2020-01-09:
Hi Michal,

Im using the one that came with the printer.

I haven't upgraded it because I read other users had issues with the newer
firmware, and I could not find in that firmware nodes mentioning this issue.

mreynolds0404, 2020-01-10:
I am experiencing  Y Axis crash during First Layer Calibration routine on 4.0.1.  Video of it happening: https://youtu.be/tiEp1syC6us

JustAnother1, 2020-01-11:
@mreynolds0404 I agree there is something wrong when doing the bed levelling. For some reason the bed does not move after the second line of levelling.

These two issues can be connected. after the "normal" bed levelling the bed would need to move completely back to start printing. Due to the levelling issue the bed is already half way there. Therefore the end stop (or Trinamic sensor in this printer) triggers much too soon. If they have some kind a plausibility check that might override the end stop detection in this case. But that is just my guesswork, haven' t checked the code.

Would be interesting if @Kasperc88 also sees this issue in bed levelling.

LuckyBenni, 2020-01-14:
I assume the Mini uses less motor current during mesh bed leveling as such lost steps also occured on the X axis several times with my Mini.

JustAnother1, 2020-01-15:
@LuckyBenni Did you try to move your axis manually? can they be moved easily? If not try to fix that! 

My understanding of the trinamic stepper drivers is that they give a feedback of how much current is needed. By adjusting the current to the feedback the motor should always turn and never overheat. But it could be that this signal does not work as well on slow moves. 

But then again why would the movement on the bed levelling need that much current?

LuckyBenni, 2020-01-15:
@JustAnother1 I tried moving the axes and apart from the motor resistance there was no special bumps or so that could be felt in both directions. I lubricated the X and Y axes with some machine oil - maybe that improves the situation a bit. But should not be the reason as my printer arrived just before Xmas and did not experience too many hours of printing.

ryanpriore, 2020-03-20:
> I am experiencing Y Axis crash during First Layer Calibration routine on 4.0.1. Video of it happening: https://youtu.be/tiEp1syC6us. Seems like something is also wrong with the second half of the mesh bed leveling (doesn't proceed beyond halfway point of bed). Checked for any physical issues and did not find any. Bed can be manually moved or via Settings through entire range without issue.

I had the exact same issue as you showed in your video. I see that you checked the physical nature of your bed. For me,  it was a seized bearing on the single bearing side of the y axis. It became worse in which the printer would try to probe off of the front of the bed so the head would crash.

JohnnyDeer, 2020-06-30:
Hi, G28 is working correct on FW 4.1.0-RC1. MBL issue seem related to HW, please make sure that Y belt is tensed right. I´m closing this issue as solved, because main problem is fixed. If someone will find out more about MBL issue please comment it here #561. Thank you all.

closed 2020-06-30 by JohnnyDeer