prusa3d/Prusa-Firmware-Buddy #17: Won't start printing

Kasperc88, 2019-12-22:
Most of the time if you plug the usb stick into the printer while its on, and try to start a print nothing happens.

Reset or turn the printer off and on lets you start the print.

michalxfanta, 2020-01-07:
We are aware of this problem and are working on a solution.

eskeyaar, 2020-01-14:
I had this issue, but after i start to eject the USB stick properly, its printing fine. Without Ejecting properly facing issue.

JohnnyDeer, 2020-06-09:
Hi @Kasperc88 , is issue still actual? Did you tried latest FW version,  if bug happens too? Good point @eskeyaar , ejecting usb stick is higly recommended.

hawwwran, 2020-06-16:
It's not fixed and the commit of @michalrudolf mentioned above seems to have no connection to this issue what so ever.

JohnnyDeer, 2020-06-30:
FW 4.1.0-RC1 have changes in USB detection. Please try if issue is still there. Thank you.

hawwwran, 2020-06-30:
> FW 4.1.0-RC1 have changes in USB detection. Please try if issue is still there. Thank you.

This problem is not present in FW 4.1.0-RC1. Confirmed.

JohnnyDeer, 2020-06-30:
@hawwwran thank you. I´m closing issue as solved.

closed 2020-06-30 by JohnnyDeer