prusa3d/Prusa-Firmware-Buddy #21: USB stick not detected, needs reboot

pleumann, 2019-12-23:
I had several cases where my USB stick would not be detected by the Mini. It's not a problem of the stick (I tried several) or the formatting (FAT32). Most of the time the specific stick even works fine. At some point it stops working and I have to restart the Mini for the problem to go away. So far I have been unable to nail it down. It could be caused by hidden metadata stuff written by my Mac or by the specific files I place on the stick (had the impression that larger GCODE files might cause the problem, maybe as part of the preview functionality).

pleumann, 2019-12-23:
Firmware 4.0.1.

Area5142, 2019-12-26:
See the same problem with firmware 4.0.1. Sometimes I can be lucky to get the print screen up after many re-inserts of the USB - often it needs a reboot. The USB stick (delivered with the Mini) works every time in the PC with Linux Mint 19.3.

mrunhart, 2019-12-26:
I am running 4.0.1 and can confirm this issue as well. Seems to occur randomly.

The provided USB stick is being properly "ejected/unmounted" from a fedora 31 system.

I think it is mainly occuring after leaving the printer idle for more than 30 minutes but am not 100% sure.

Rebooting has had a 100% success rate at fixing the issue for me. both with, and without the usb stick being installed at boot.

drdelaney, 2019-12-28:
Can confirm this same issue with 4.0.1.  It seems to happen when I remove a USB stick/USB SD card reader that is not the official prusa USB stick.  Once it stops working, even the prusa USB stick stops as well.  Until a reboot, then both work.

GitSteeK, 2019-12-28:
Can confirm, too. After a long printing-pause (30min) I ejected the stick an plugged it in again after I saved new files on it and the stick wasn't recognized. Maybe a timeout related problem?!

michalxfanta, 2020-01-07:
We are aware of this problem and are working on a solution.

freshmas, 2020-01-12:
I feel like this hasn't happened again ever since I made sure to only remove the USB drive when the printer was on the home screen.

Don't know if it's relevant and not thoroughly proven. Just a theory.

pleumann, 2020-02-22:
Haven't had that problem again since updating to 4.0.3. USB stick is always detected, both during boot and when re-inserting while the printer is on. Thanks for fixing this!

soinwi, 2020-04-06:
I'm not sure if my USB-related problem is the same issue as the one mentioned here. But for me it does not recognize the USB-stick if there are no printable files in the main folder. (I only have printed files in the subfolders, so if I plug in the stick it is not recognized. Rebooting or adding any random printable file to root dir makes the mini recognize the stick.)

Pitel, 2020-04-06:
It might be very related. I also don't have any gcode in the root of the drive, only in subfolders.

Pitel, 2020-04-07:
When I put gcode file in the root of the drive, it pop ups after inserting the drive. So I still have to press back, and select the file from directory I want.

So, that's a change. But not very helpful.

But it should definitely help the devs to narrow down the problem, because it gives them specific scenario to test. :+1:

0xygen8, 2020-04-07:
What type of USB stick did you use: USB3 with Blue or USB2 with Black inside -- when I switched from USB3 to USB2 sticks all my problems went away.

Pitel, 2020-04-07:
USB2. The metal one provided with the printer. Also using short USB2 extension cable for batter access to the port.

0xygen8, 2020-04-07:
Have you tried without the extension cable?

Vic-Y, 2020-04-11:
Currently running 4.0.5. I have the same/similar symptom of the Print item on the Home screen saying No USB underneath when I insert a USB stick with no g-code in the root directory. 

However, I found that I don't need to reboot. If I select one of the other Home screen items and then immediately return to the Home screen, the Print button is then active/works. So the problem may be with the UI code and not with the underlying USB drive communication code.

JohnnyDeer, 2020-06-22:
Hi, sensitivity of usb stick detection is improved in 4.1.0-RC1. Please try reproduce this issue on newest FW :)

JohnnyDeer, 2020-07-21:
No reaction, I´m closing as solved.

closed 2020-07-21 by JohnnyDeer

Vic-Y, 2020-07-22:
Sorry for the slow response. This problem is solved for me in 4.1.0. Thank you.