prusa3d/Prusa-Firmware-Buddy #22: Print not being shown as completed

pleumann, 2019-12-23:
I had several prints that did not "end" properly, meaning the Mini would be in "pause" rather than saying the print was successful. First I though it might have beeen caused by a print that was exactly 18 cm in height, but then I saw the same for smaller prints, too.

pleumann, 2019-12-23:
Firmware 4.0.1.

AndyThirtover, 2019-12-23:
I have seen the same on all my prints since 4.0.1 -- have to manually mark them as complete.

ryanpriore, 2019-12-24:
This is the same for me as well with pre-sliced files from the USB stick or files that I have sliced.

Area5142, 2019-12-27:
Firmware 4.0.1
Same problem here. Printer just waits after print is finished. Have to select stop to end print.
Tried a simple G-code file with just a single G-code "G0 Z100" and some comments - same result, it executes and just waits.
Looks like the firmware never detects end of file (EOF).

LuckyBenni, 2020-01-14:
Same here - but not with all prints though :-(

JohnnyDeer, 2020-06-09:
Hi, this bug is fixed in 4.0.5., i'm closing this issue.

closed 2020-06-09 by JohnnyDeer