prusa3d/Prusa-Firmware-Buddy #23: Mesh Levelling has very weird behaviour

AndyThirtover, 2019-12-23:
4.0.1 Please view:

https://www.youtube.com/watch?v=rOWLI0K9chY

You'll see that the points 1-2 work perfectly and the 3-4-5 only go UP.  6-7-8 are ok and then

9-10 do the same thing (as do all of the points on the left hand side of the bed.

The net result is that this printer cannot be first layer calibrated.   It seems that the issue gets worse every time I power cycle/factory reset the printer.    I've managed one print on 4.0.0 but since the 4.0.1 upgrade this sad Prusa Mini has not managed a single print ;)

lennertvdkort, 2019-12-23:
On one of the FB-groups there was someone that had the same problem.
One of the wires of his Minda is broken.
When extruder goes to the end the strain on the wires is to big and then the broken wire doesn't make contact anymore.
This is IMHO a big design flaw, the wire bundle almost makes a 90° bent when the extruder is at the end. Mine also is that bad(my mini has a skewed bed and my smooth plate is already worn because of that on one side.. Not a good start)

AndyThirtover, 2019-12-24:
@lennertvdkort I agree.   I have taken the cable tie out and I can see that the MINDA LED flashes as I wiggle the cable.   Will request a new one after the Christmas break -- which starts now!

DooMon, 2019-12-26:
Same problem here! 1-3, 5-7, 11 are ok, the rest MINDA goes only up.

michalxfanta, 2020-01-07:
This is not a firmware error, it is probably caused by broken MINDA cables.