prusa3d/Prusa-Firmware-Buddy #24: Thermal Runaway Bed-error when placing steel sheet on heated bed

6ffm70, 2019-12-25:
**Description**: When the printer is pre-heated w/o spring steel sheet, and the cold (~ambient temperature) steel sheet is placed on the printer bed, the steel sheet cools the heated bed down by an amount of Kelvins (visible in the display before the warning).
The printer then stalls with the error message 

> THERMAL RUNAWAY
> Bed
> [blank lines]
> 4.0.1-final+1974

**Possible functional issue**: the firmware is specified that this temperature drop is not plausible, and assumes the bed sensor is giving false readings.

**Proposed solution**: allow for higher temperature drops in the firmware to allow simultaneous pre-heating and cleaning/preparing bed for the next print.

Area5142, 2019-12-26:
Had this error a couple of times too, when preparing the print-bed. Only way out is reset.

6ffm70, 2019-12-27:
Tried a workaround to avoid having to reboot the printer and preserve as much heat as possible for the next print (to avoid waiting for another temperature ramp-up):

**Description**: After removing the steel sheet immediately after a successful print to prepare it for the immediately following print, I set the printer to cool down, then placed the prepared sheet again, and then set the printer to preheat. After a couple of seconds, the printer stalls with the error message (output in Octoprint - not in the display)

> (Error: Heating failed, system stopped! Heater_ID: 0 - Printer halted. kill() called!)

**Possible functional issue**: the firmware still recognises a temperature drop, if preheating is switched on while the colder steel sheet cools down the heatbed.

michalxfanta, 2020-01-07:
We are aware of this error and are preparing a solution.

closed 2020-01-07 by michalxfanta