prusa3d/Prusa-Firmware-Buddy #26: Default is to re-print after print?

Area5142, 2019-12-26:
95% of the time, I only print a model file once. If I print it more than once, it is properly in another material or colour.

When a print finish, the default predefined action is to re-print the model. I often, by mistake, click the button and starts a new print of the same model. To get out of the situation, I have to stop the print and confirm. The heaters has properly been turned on as well and must be turned off...

In my mind the right default action would be to return to the main menu or the file list.

If I had a print farm, this option would be nice to print a lot of the same parts, but this printer is recommended to beginners and schools. Any default action should always be harmless in my opinion.

drdelaney, 2019-12-28:
I agree with changing this to default to home.  And/or adding in a timeout to go to home after a certain time if nothing has been chosen.

michalxfanta, 2020-01-07:
Thanks for the idea, we'll consider change default action after finished print.

closed 2020-01-07 by michalxfanta