prusa3d/Prusa-Firmware-Buddy #27: X-axis crash

pleumann, 2019-12-26:
I just had a case where I needed to cancel a print during the first layer. I think it was still in the skirt. The x-carriage crashed into the right side (well, maybe that's a harsh word, but it obviously did not find the endstop). Single case so far. Firmware 4.0.1.

Area5142, 2019-12-27:
Firmware 4.0.1
Stopped print in third layer as the model had gone loose. The print head lifted up and x-carriage moved right and crashed into end stop. Motor did not stop on end stop, but was running for about a second more, making a clicking noise.
I have only seen this happen once.

GitSteeK, 2019-12-28:
Same here... stopped printing in one of the first levels an the print head lifted up and crashed into the end stop ... terrible noise ans white powder fall on the heatbed.

pleumann, 2019-12-29:
I just had a case where the same happened for the y-axis. Again, after canceling a print. So maybe this issue can be merged with #16.

Lightn, 2020-01-01:
I had this happen twice at the start of printing the sample tree frog included on the USB drive.  At least I think it was the X-axis, I hit reset pretty quick when it happened.

laposheureux, 2020-01-05:
Have also had this happen with 4.0.1 on the x-axis. It was after I'd canceled a print as well. It's like the printer loses track of where the print head is upon cancellation and then also fails to detect when it's reaching the end.

michalxfanta, 2020-01-07:
We try to replicate this behavior.

freshmas, 2020-01-09:
I have also had this happen at least 3 times after cancelling different prints. Most recently, a clump of white, powdery material fell off just like GitSteek above.

LuckyBenni, 2020-01-14:
I also experienced this behaviour several time at my Mini with 4.0.1.
It seems that the motor which moves the extrusion carriage from left to right uses too little power which results sometimes already during the mesh bed leveling process in lost steps. Sometimes I noticed that between the first and second measurement point, sometimes between the second and third measurement. Apparently at the second row the carriage then crashed to the Z axis.
After rebooting the Mini and running the exact same gcode the mesh bed levelling succeeds w/o issues.
If it is not related to too little power another reason could be electronic interferences - but they would be a nightmare to find out :-(.

michalxfanta, 2020-01-24:
We replicated this bug, we will fix this asap.

x10102, 2020-02-06:
Also happened to me
FW 4.0.1

Jivens75, 2020-03-07:
Same here
FW 4.0.4

JohnnyDeer, 2020-06-30:
Hi, this issue should be fixed on 4.1.0-RC1. Please try it out :)

JohnnyDeer, 2020-07-28:
No responses. I'm closing this issue as solved.

closed 2020-07-28 by JohnnyDeer