prusa3d/Prusa-Firmware-Buddy #30: Filament tip problem

gdenicolas, 2019-12-27:
After finish printing, and some hours later, the filament tip gets cold in a form that could not go through the ptfe tube when I was trying to unload the filament.  To be able to unload the filament, I unscrew the brass nut and cut the filament tip, then the filament was easily unloaded.

Maybe you can add some filament movement right after ending printing to change the filament tip.

wdgeek, 2020-01-02:
I have a similar problem.
I have found that if I run a print and dont unload the filament immediately after the print finishes I cant just do a unload filament. I will get a filament tip that is too large and it wont pass through the PTFE tube as it exits the hot end. The comments about the unload process doing a "reverse" ramming seem to be correct since when unloading the Mini seems to first "retract" the filament (I can see some filament reverse out, and then it tries to do the ram and unload, but I think in the first "reverse" ramming of filament causes a bulb to form at the tip of the filament and after that the filament gets stuck and it gets "stripped" at the filament feed gear since the filament can no longer pass through the PTFE tube. This happened every time I try to do a unload AFTER the printer has been idle for some time. This is a pain to try to un-jam because you have to take apart the PTFE tube connections to try to get the filament unstuck. Sometimes I have to take out both sides of the connections to try to unstick the filament.  It unloads properly when I unload immediately after the print finishes.
HOWEVER, I found that if the filament has been left in the printer for a long time then I found that if you FIRST do a "Purge" filament, then immediately after the purge do an "Unload" filament, then it has always unloaded the filament for me without problem and with a nicely formed tip that easily passes through the PTFE tube.
I hope Prusa tries to do some sort of fix for this since its kind of irritating to have to do a purge then unload.
BTW, I don't understand why the Prusa MINI doesn't have an option to just unload the filament right after a print. I assume that they are keeping the filament in the MINI so reprints are easy, or they felt that in a production environment filament changes for every print are not common. But unless they fix this unload filament problem it will remain a real irritation to have to do a PURGE followed by an Unload if your printer has been idle for some time. (I usually run a print and when I go to check on the print it would have been a long time since the print finished, so for me its very common for me to not unload the filament right after I print)

michalxfanta, 2020-01-07:
We are working on a new filament unload sequence to solve this problem.

closed 2020-01-07 by michalxfanta