prusa3d/Prusa-Firmware-Buddy #40: Change filament assumes "yes" on second check

drdelaney, 2019-12-28:
I mis-loaded my mini a few times, and unlike the mk2.5s/mk3s where it keeps asking if the filament is the right color, the mini only asked once, then assumed "yes" on the second check and started to print right away.

There was clearly no filament going down to the hotend.
I would suggest enabling a check loop to keep asking if it is the right color.  And perhaps if the 2nd or 3rd answer is "No", ask if the filament was loaded correctly.
If yes, keep asking for color, or possibly cancel (who needs to hit it that many times), but if no, start the load cycle over again, so the filament can be re-inserted back down the bowden.

I lost a print an hour in, because I ran out of filament, and mis-loaded the new filament (I pushed in as far as it would go, but it was not grabbed)

Area5142, 2019-12-28:
Related to: Load Filament and Color Change Bug #15

drdelaney, 2019-12-28:
> Related to: Load Filament and Color Change Bug #15

Here I go trying to be helpful and I messed that up too. 😅

michalxfanta, 2020-01-07:
We are working on adjusting the filament loading behavior. The fix will appear in one of the other releases.

closed 2020-01-07 by michalxfanta