prusa3d/Prusa-Firmware-Buddy #41: Starts to print something when USB is removed at the print finished screen

drdelaney, 2019-12-28:
After a print is finished, if one does not select either reprint or home, when the USB is removed, "something" tries to print, as the Z axis moves up, and the buttons change from "reprint" and "home" to "resume" and "stop".

The heaters do not seem to activate as they show the current temp over 0.

If you click stop, the printer thinks for a second, then lowers Z back to where it was, or close to where it was.

drdelaney, 2019-12-28:
Tried a gcode with no movements and it did the same thing, minus moving the z axis. It started some sort of print. I think it might be somehow activating the reprint option, which stops when the USB is removed.

laposheureux, 2020-01-05:
Have noticed this as well - gotta fully exit the print finished screen before unplugging USB or 100% of the time it goes into this "thinking it is printing" state.

michalxfanta, 2020-01-07:
We already fix this issue and will be released in the next version.

closed 2020-01-07 by michalxfanta