prusa3d/Prusa-Firmware-Buddy #43: HardFault on access to Web-Control

Luzifer, 2019-12-29:
- Firmware: `4.0.1-final+1974`
- Browser: Firefox 71.0 on Linux

Steps to reproduce:
- Boot up printer
- Let an IP get assigned
- Open `http://<ip of printer>/` in browser
- Experience crash and fan spin-up on printer while getting only a partial website

Note:
- In **some** cases the site gets loaded without image but with stylesheet
- Most times the site looks like displayed below
- After reset on crash the (broken) website can refresh temperatures without issues
- Possibly related to / duplicate of #18 but in my case not related to an active print but occurring after freshly starting the printer without any print task

Trace:
[image]

Website:
[image]

Lichtjaeger, 2020-02-11:
Perhaps I should also mention that a Jumper was sent after me. I haven't put it in because I don't use custom firmware. Is it possible that I still need to place it somewhere?

michalxfanta, 2020-02-17:
We tested it and yes it's known bug.

JohnnyDeer, 2020-06-09:
Hi, bluescreen bug is already fixed on newest versions of firmware, i´m closing this issue.

closed 2020-06-09 by JohnnyDeer