prusa3d/Prusa-Firmware-Buddy #44: Unload filament would not heat/start

pleumann, 2019-12-30:
I just had a case where I confirmed unloading the filament, but nothing happened. The Mini did not heat up, the display says "ramming", but no motors were moved at all. This happened immediately after I had to manually finish one of those seemingly unfinished print jobs (see #22). Maybe that's a connection to look into. Firmware 4.0.1.

michalxfanta, 2020-01-07:
We try to replicate this behavior.

pleumann, 2020-01-19:
I had this a couple of times again. It may be tied to either of these or a combination of the two:

1) Pulling out the USB stick after a print.

2) A finished print job not being flagged as finished, as described in #22.

I think it can also appear when starting a print. Usually a quick reset is the easiest workaround.

michalxfanta, 2020-01-24:
We replicated this bug, we will fix this asap.

JohnnyDeer, 2020-06-09:
Hi, this bug is fixed in latest FW version., i'm closing this issue.

closed 2020-06-09 by JohnnyDeer