prusa3d/Prusa-Firmware-Buddy #49: backing out of the initiated program

PrintingPez, 2020-01-03:
(understand that this is my first 3D printer and also my first experience with a 3D printer) I set up my Prusa Mini but struggled to backtrack once I started setup. The reason I wanted to backup was two-fold:
the filament wasn't feeding into the long white tube that is attached to the print nozzle...had to unplug the machine to try again (numerous times until the feed-gear finally grabbed the filament and fed it to the print nozzle....no apparent reason this was so difficult to feed the filament)
when the test part became separated from the heat bed, I could not interrupt the printing without again unplugging the machine.
If I had a way to interrupt the setup process in either instance, I'm certain it would of been better than unplugging the power supply.
Thank You Josef for requesting feedback from new owners.

michalxfanta, 2020-01-07:
When inserting the filament, make sure that the end of the filament is properly cut to the tip.
[image]
Loading filament is discussed in chapter **6.6 Loading and unloading the filament** in our [3D Printing Handbook](https://cdn.prusa3d.com/downloads/manual/prusa3d_manual_mini_en_1_0.pdf)

closed 2020-01-07 by michalxfanta

PrintingPez, 2020-01-07:
>   CAUTION: This email originated from outside your organization. Exercise caution when opening attachments or clicking links, especially from unknown senders.

Thank You...the filament I loaded was brand new from Prusa and it appeared to of been cut correctly from their factory, so I am still troubleshooting the filament feeding issue I've experienced....nice of you to send the photographs for proper tip angle

Sent from HCL Verse

   Michal Fanta --- Re: [prusa3d/Prusa-Firmware-Buddy] backing out of the initiated program (#49) ---