prusa3d/Prusa-Firmware-Buddy #72: Bed Level Correction needed

freshmas, 2020-01-09:
I can't get a good first layer on large prints. The majority of the bed prints nicely on both textured and smooth sheets; however, the back-right corner fails to stick. If I adjust down with live Z to get the back-left corner to stick, the rest of the bed is too squished.

After working with live chat support, there does not seem to be an easy mechanical solution in this case. If this was happening on my MK3, I believe it could be easily remedied with the Bed Level Correction function.

koenkooi, 2020-01-09:
I had a similar issue and found that in my case the MINDA was affected by temperature differences. 
The 'tilt' matched the path of the bed leveling, so I changed the start gcode in prusaslicer to be:

> G90 ; use absolute coordinates
> M83 ; extruder relative mode
> M92 E317 ; set steps/unit for extruder
> 
> **G28 ; home all without mesh bed level
> M140 S[first_layer_bed_temperature] ; set bed temp
> G1 X115 Y95 Z2 F9000 ; Center MINDA on the middle of the bed 
> M190 S[first_layer_bed_temperature] ; wait for bed temp**
> 
> M104 S170 ; set extruder temp for bed leveling
> M140 S[first_layer_bed_temperature] ; set bed temp
> M109 R170 ; wait for bed leveling temp
> M190 S[first_layer_bed_temperature] ; wait for bed temp
> G28 ; home all without mesh bed level
> 
> G1 X115 Y95 Z2 F9000 ; Center MINDA on the middle of the bed 
> 
> G29 ; mesh bed leveling 
> M104 S[first_layer_temperature] ; set extruder temp
> G92 E0.0
> G1 Y-2.0 X179 F2400
> G1 Z3 F720
> M109 S[first_layer_temperature] ; wait for extruder temp
> 
> ; intro line
> G1 X170 F1000
> G1 Z0.2 F720
> G1 X110.0 E8.0 F600
> G1 X40.0 E10.0 F400
> G92 E0.0
> 
> M221 S95 ; set flow

This warms up the MINDA *first* so it stays the same temperature during probing. After that change I didn't have issues with a 'tilted' first layer anymore, or with wildly varying Z-offsets.

michalxfanta, 2020-01-09:
We will certainly look into improving Mesh Bed Lelling, but as suggested by @koenkooi preheating the MINDA sensor can improve the result.

closed 2020-01-09 by michalxfanta

ProdYJ, 2021-04-06:
Hi,

I have the exact same problem with my mini (back right corner not sticking on large prints), even with a preheat. Moreover, I still have the problem after I have installed the Super Pinda. Any hope to see a bed level correction implemented soon ?