prusa3d/Prusa-Firmware-Buddy #82: Allow for faster baudrates over USB serial port

koenkooi, 2020-01-10:
Streaming gcode with lots of tiny arc segments over serial is still troublesome, a baudrate of 115200 just isn't enough for that. Allowing for e.g. 230400, 460800, 921600 or the atmega special 250000 would be a nice feature.

JustAnother1, 2020-01-11:
Did you actually have an issue with to slow data transmission or do you just want to so a higher value on your PC?

As you can see [here](https://github.com/prusa3d/Buddy-board-MINI-PCB/blob/master/rev.1.0.0/BUDDY_v1.0.0.pdf) the USB is connected directly to the MCU. So the baud rate is a lie. There is no UART involved. As long as your PC does not limit the data rate to the baud rate (it should not do that) the data can be sand as fast as the MCU can handle it.

koenkooi, 2020-01-11:
First, I'm not using a PC. Second, I see blobbing and other issues on the same gcode printed over serial compared to printing it from USB.
Also, at least on the linux side the baudrate you set on such a 'software' uart does get used.

JustAnother1, 2020-01-11:
I assume with "serial" you mean the Universal Serial Bus(USB) and not some RS232 or UART.

I would suggest to just set the baud rate to whatever you desire on your not PC Linux machine (RasPi?) It will probably just work fine.

bkerler, 2023-12-15:
This issue should be closed. As JustAnother1 already addressed, setting a baudrate doesn't make any sense for the usb serial port, that's why it's also disabled as M575 command. @koenkooi if you mean a direct connection to the board using real uart, please change the issue topic accordingly.

closed 2024-04-13 by CZDanol