prusa3d/Prusa-Firmware-Buddy #84: Calibrating "Mesh Bed Level." no longer works on my Prusa MINI

cyrozap, 2020-01-11:
Hello, I've been successfully printing on my MINI for a little over a week now, until today when I had to remove some stuck filament from my extruder and the automatic bed leveling stopped working.

What happens is, when I start "Mesh Bed Level." from the Calibration menu:

1. The Z-axis auto-homes.
2. The extruder moves to the first point, goes down and up, with the LED on the sensor blinking.
3. The extruder moves to the second point, goes down and up, with the LED on the sensor blinking.
4. The extruder moves to the third point, but sometimes instead of going down and up, it goes up twice, and the LED never blinks.
5. The extruder moves to the fourth point, and always just goes up twice, and the LED never blinks.
6. This process repeats for the next three lines, with the points on the X-axis closest to the Z-axis calibrating properly, while the points furthest away fail to calibrate at all.

I'm not sure why this is happening--my MINI appears fine, nothing seems misaligned, and the screws are properly tightened--so I'm not sure what I need to do to fix this, if there's anything that can be done without modifying the code. If code changes are needed, I'm not opposed to doing the work myself, but I'm not familiar with this code base or how to get started developing and debugging on this platform.

In my opinion, a minimum fix would be to error-out if the z-axis moves up twice without the sensor LED blinking (to avoid wasting the user's time), but a better fix would be to never move up on the first move in the first place, and the best fix would enable the user to "guide" the extruder head to the proper height when an error is detected (so it doesn't crash into the build plate) and then re-do that point's calibration.

This is happening on the latest firmware (`4.0.1-final+1974`).

JustAnother1, 2020-01-11:
If you want to started in working with the code base look at the readme and clone the repository. See if you can get it compiling. If you can not get it to build please complain.

If you want to let your code run on your printer then you have to also mechanically destroy part of your electronics. Prusa describes how to break away the thing that blocks non signed firmware.

On your issue: Other users have complained about to tight zip ties on the pint head cabling. When the print head is closest to the Z axis the bending in the cables going to the print head is the highest. This leads me to assume that in your printer the cable for Z-Dir might be broken. This would explain the moving up instead of down. You can test this assumption by sending commands to move the head up and down in close proximity to the Z axis. Octoprint can do this for example.

freshmas, 2020-01-12:
It sounds like maybe the sensor is being triggered in that X position. If I wasn't printing at this time, I would do a test by holding a piece of metal under the sensor and see how the machine behaves.

jokohoko, 2020-01-13:
I have the exact same issue since a moment ago.
I am certain the minda probes cable has broken.
Do the following test: When you move the Z up, the led in the top of the sensor should be lit up constantly.
when you now move the print head slowly left to right in my case the led went out and back on. Proving the cable if faulty.
I just had contact with prusa support trough chat and they are sending me a replacement sensor.

michalxfanta, 2020-01-16:
The problem described is likely to be caused by damaged cables at the MINDA sensor. Contact our [support](https://help.prusa3d.com/article/6dY1xjal2N-support)

closed 2020-01-16 by michalxfanta