prusa3d/Prusa-Firmware-Buddy #98: Error in minda probe is not detected by mesh bed leveling

jokohoko, 2020-01-13:
I have a broken cable on my minda probe and thereby noticed a bug in the firmware.
When the sensor fails to detect the bed on the MK3 it gives a warning and re-calibrates.
When the sensor fails on the Mini by not giving a response, the head goes up by some millimeters and goes on to the next location. When it starts to print the head goes up by the same few millimeters on these points.
So when hitting the maximum height allowed in the calibration it should give a warning!

jokohoko, 2020-01-13:
error report #84 is about the same issue. But i see it as a firmware issue where he focuses on the mechanical error of the broken cable.

michalxfanta, 2020-01-16:
You're right, we're working to improve Mesh Bed Leveling and detect a damaged MINDA.

closed 2020-01-16 by michalxfanta