prusa3d/Prusa-Firmware-Buddy #108: Heatbed positioning screws impacting mesh bed leveling procedure?

LuckyBenni, 2020-01-14:
When observing the mesh bed leveling procedure on FW 4.0.1 it seems that in the last row of measuring points (at the very back of the bed) the second and third measurements near the heatbed positioning screws result in less Z axis movement compared to all the other measurement points.
If that is a reproducible "problem" I assume this is already reflected in the firmware???

michalxfanta, 2020-01-16:
Is the MINDA sensor in the last row OK (sensor is blinking)?

LuckyBenni, 2020-01-20:
Yeah - the MINDA sensor is okay. it just seems that for some measurement points the motor needs less time to measure then for others - especially when there are the screws at the back around,

Another observation I just had: is it normal that the adjustment height needs to be changed between standard PEI sheet (which is approx. 1.7mm thick) and the powder coated sheet (which is approx. 0.7mm thick)? With my powder coated bed I need to use a value around -1.25mm to -1.35mm to get the material to stick well, with the original PEI sheet I need to use -0.9mm to -0.95mm, which is a bummer if you change the beds the first time and your nozzle almost damages the PEI surface. Just wanted to know whether this is normal or may indicate an issue with the MINDA readings ...

michalxfanta, 2020-01-22:
Thank you for the video, I'll forward it to our developers.

JohnnyDeer, 2021-08-10:
Hi @LuckyBenni, bed position screws do not have any impact to MBL, MINDA or extruder itself. We never observed any complications or influences and this report is the only one that deals with this topic. Please consider closing this issue.