prusa3d/Prusa-Firmware-Buddy #136: Printer is not setup to work properly with 0.25mm nozzle

laposheureux, 2020-01-20:
While PrusaSlicer has configuration options for it, the Mini itself does not have any mechanism I can tell to compensate for a nozzle change. I've gone back through the initial wizard and first layer calibration and all the menus I can find and there's nowhere that I can tell it to not expect the ability to extrude for a nozzle other than 0.4.

As a result:
- First layer calibration can't complete as it begins to ram/clog the filament at the end of the nozzle near the end
- Purge commands try to push too much filament through and can cause the extruder gears to grind away the filament until it cannot push any further
- Loading filament is the same as above

laposheureux, 2020-01-21:
Issues above aside, I am able to get it to print quite competently with the nozzle using PrusaSlicer settings set to the 0.25mm nozzle, it's just all the on-printer setup that is a problem.

michalxfanta, 2020-01-30:
We plan to add a nozzle diameter setting in the printer firmware so that the First layer calibration and other features take this setting, just as we have on the MK3.

x10102, 2020-02-18:
> We plan to add a nozzle diameter setting in the printer firmware so that the First layer calibration and other features take this setting, just as we have on the MK3.

Any idea when will it be implemented?

JohnnyDeer, 2020-07-28:
Hi, we don't plan to implement this feature in near future. I personally print with 0.25 nozzle for long time without any problems, nothing blocking the use :)

handro123, 2022-12-04:
I think this should be re-visited with the release of the E3D Revo system. Being able to do a first layer calibration for different nozzle sizes built into the printer would be nice to have.

Prusa-Support, 2023-01-06:
Thank you for your suggestions.

The ability to print a "zig-zag line" (pre-loaded _first layer calibration_ prints) that automatically adapts to the nozzle size (_HW setup_) would surely be a handy feature.

Our developers will make their considerations about this.

At the moment, our suggestion is to slice a simple print file on your own, as per the following instructions.
https://help.prusa3d.com/article/different-nozzle-types_2193#first-layer-calibration-of-other-nozzles
After all the "zig-zag line" is only a quick way to set a rough Adjust Z value. In most cases, you may want to print a squared, full, regular first layer (infill + perimeters) to further fine-tune your value.
In Prusa Slicer, you can "_add shape_" of a cube, and scale it 50x50x0.2 mm in a few clicks only, or print and pause any other suitable object to evaluate the first layer.

Michele Moramarco
Prusa Research

closed 2024-07-04 by github-actions[bot]