prusa3d/Prusa-Firmware-Buddy #145: crash recognition at mesh bed leveling is not working, printer will destroy itselfe

pinguforce, 2020-01-22:
firmware 4.0.1 offizial Prusa Driver site release.

yesterday i forgotten that i took of the spring steel plate and started a print. a few minutes later i looked after it.

the nooze crashed so forcefull into the hotbed, that it got scrached very deep (you can scratched and bentded cooper), bented about 3 cm and even remained a massive dent. have to try out this evening if its still usable. i think i need a new hotbed :(

so be carefull, crash recognition on the mesh bed leveling does not work!
THIS SHOULD BE DEFINITLY BE HOTFIXED

michalxfanta, 2020-01-30:
In the next firmware version, there will be an improvement in the Z-axis crash detection, which should avoid such a situation. Contact our support for a heatbed replacement.

x10102, 2020-02-06:
I have this bug too, it destroyed my PEI sheet. This needs a fix ASAP!!!

zoltan-l, 2022-01-14:
The bug is fixed, At the print start when the first probing fails - the nozzle gets lower than allowed the print stops and reports the Probing error.
If the user intentionaly remove the MBL code from GCODE and removes the sheet from bed is on his/her own risk

closed 2022-01-14 by zoltan-l