prusa3d/Prusa-Firmware-Buddy #149: MMU2S not supported

HutyJarl, 2020-01-23:
it is not possible to use MMU2S in mini prusa for a software problem says a guy from PRUSA support.

how can we go around the problem?
Prusa's technical support adds that even other MMs (multi materials) are not compatible.

HELP BOYS :(

Is anyone trying to work on it?

JustAnother1, 2020-01-25:
Did you damage your board? Only people that damage their electronics and therefore loose the warranty can test firmware that is not signed by Prusa.

Lichtjaeger, 2020-02-02:
MMU2S can only work with a direct extruder with integrated filament sensor.
The Mini has a Bowden extruder.

But you can use [Palette 2S](https://www.mosaicmfg.com/products/palette-2s) if you need multi-colour/multi-material.

HutyJarl, 2020-02-03:
palette2s is the best solution for me

to use more colors I saw that you also have an application that manages the
manual color change. it is correct ? . is it compatible with MINI?

HutyJarl, 2020-02-03:
you.. ahaha PRUSA*

Lichtjaeger, 2020-02-03:
> to use more colors I saw that you also have an application that manages the manual color change. it is correct ?

If you are referring to "colour change by layer": This is supported by every printer. You can add the colour steps during slicing.

HutyJarl, 2020-02-03:
perfect :))
thank u

closed 2020-02-24 by michalxfanta