prusa3d/Prusa-Firmware-Buddy #157: Nozzle cooling to 170°C during probing not consistently implemented

6ffm70, 2020-01-26:
**Description:**
When I preheated the printer for any material, and start a print after that temperature has reached, the f/w first cools the tool down to 170°C, then proceeds mesh bed leveling with that temperature. Once complete, it heats up to print job temperature and prints as expected.

When doing first layer calibration, the f/w will not cool down to 170° tool temperature, but proceeds mesh bed leveling with the selected material temperature. Both cases are reproducible.

**Assumption:**
Cooling the tool to 170°C is required on the Mini, because the MINDA probe does not feature a thermistor for heat compensation.

**Expected behaviour:**
For the same reason the tool is limited to 170°C during pre-print mesh bed leveling, the tool should be limited for pre-1st-layer-calibration mesh bed leveling as well.

michalxfanta, 2020-02-24:
You are right, the behavior should be the same in both places.

loglow, 2020-03-02:
I just want to add my experience to this issue. I've had some problems so far with getting the first-layer height offset calibrated. The correct value for my printer seems to drift around. A particular Z-offset value that was perfect for the last print will suddenly be either too small or too large on the next print, and I wonder if this drifting has something to do with the (lack of) MINDA probe temperature compensation and/or with this particular open issue?

keksa, 2020-05-05:
@michalxfanta If the cooldown to 170 °C is required due to MINDA not having thermistor, this should be IMO marked as a bug and not just enhancement.

I don't have the technical knowledge for this, but it sound like the mesh bed leveling will be incorrect when done with higher temperature and thus it could result in wrong first layer calibration? There is also the secondary problem of tiny filament blobs all over the place.

P.S. still consistently reproducible on FW 4.0.5, just run the first layer calibration from calibration menu, MBL will be done with 215 °C nozzle temp with PLA

JohnnyDeer, 2020-06-24:
Hi, FW version 4.1.0-RC1 solving this bug. Please try it out. I´m closing this issue as solved.

closed 2020-06-24 by JohnnyDeer