prusa3d/Prusa-Firmware-Buddy #189: "feature request" show information's of prints from OctoPrint

Strankler, 2020-02-09:
**Description:**
While prints initiated from OctoPrint, the display shown no progress of the print, but only the home screen (plus temps, nozzle height, speed ratio and material).

**Feature request:**
Show same information's in the display for prints initiated from OctoPrint as for print jobs from USB drive, at least progress in percent, printing and remaining time, like the MK3 does.
Also the option to stop or pause the print would be great.

michalxfanta, 2020-02-24:
Thank you for an interesting idea, we will consider its implementation.

fawick, 2020-02-24:
I second this feature request.

In particular, it would be great to see support for the following codes when printing via USB cable from Octoprint
- M73
- M117

steve220, 2020-03-01:
I'll third this request :-)

6ffm70, 2020-08-01:
I fourth this request

yonie1, 2020-08-27:
+1 this would be really helpful and add value and the code to display is already there because it can display it when printing from USB.

naltatis, 2020-10-22:
A year ago I've built this octoprint based flexi-rex progress indicator with an e-paper display on a raspberry pi. I'd love to see the magic octopus slowly getting filled on the Prusa mini screen.
[image]

mucek4, 2021-01-07:
Any news on this? I really miss that information and a picture of octo is useless.

gimpfenlord, 2021-02-04:
i also "upvote" this.
currently my LCD is useless 99% of the time. so much unused potential

Joshi425, 2021-02-17:
this would be awsome please @michalxfanta

w90, 2021-04-12:
Great idea! Upvote!
Now in order to see the status of my print, I need to visit the octoprint.local page in browser

LimeAyy, 2021-04-16:
Here to also request this.

aguschanchu, 2021-04-19:
Please implement this!

Gordontgopher, 2021-04-26:
What does this give you that setting "Action Command Prompt Settings" to 'never' in Octoprint doesn't give you?

thefreewheeler, 2021-04-30:
Any news on this? Hugely disappointing that the Mini's LCD shows no useful info.

matthewlloyd, 2021-06-28:
Anyone following this issue may be interested to know that I just released custom firmware that adds support for `M73` (print progress) and `M117` (LCD messages):

http://github.com/matthewlloyd/Llama-Mini-Firmware#readme

coaxial, 2021-07-01:
This is sorely missing. The Mini LCD is so large, why take all the space up with an Octoprint logo? It’s cumbersome to go and find a computer or phone and visit the Octoprint page just to find out how far along the print is…

redPanther, 2021-07-02:
It's a pitty that we have a nice screen with so little information. In the meanwhile I have a workaround:
if octoprint is used, install DisplayLayerProgress Plugin. I use this for printer message: `[progress]% L=[current_layer]/[total_layers]`

then when printing, select "tune" scroll all the way down to "messages". Then you see the progress in % and current and max layer. Temperatures and z height is already displayed at the bottom. Looks absolutely not so nice as it could, but it provides necessary information during print.

coaxial, 2021-07-02:
> Anyone following this issue may be interested to know that I just released custom firmware that adds support for `M73` (print progress) and `M117` (LCD messages):
> 
> [http://github.com/matthewlloyd/Llama-Mini-Firmware#readme](https://github.com/matthewlloyd/Llama-Mini-Firmware#readme)

Would love to try your firmware, but really not keen on voiding my warranty just to get M117 and M73 support. And there is no other way at the moment, I think.

theHalo, 2021-08-10:
Was hoping this would have been included in today’s release. Bump.

pat1121, 2021-08-17:
+1 would also like to see this!

coaxial, 2021-08-17:
I have a PR (#1497) that would enable this based on the work done for the llama firmware but its not building for some reason (can't see why.) I wish someone more comfortable with C could give feedback and help pinpoint what's going on, because if the build for that PR is green then it has more chances of getting merged.

mtaumike, 2021-11-11:
Bumping this, however it gets done.

Be fun to get some progress for when walking up to the printer, and even better, if any thumbnail that was created in the gcode file could also be shown vs 'the octopus'

atxkev, 2021-11-28:
+1 to add this in. My MK3s screen with octoprint is more useful.

Alshootfa, 2022-01-07:
+1 will allow us to take advantage of the Prusa Mini's awesome color display instead of covering it up.

Aelirenn666, 2022-01-09:
I have recently started using OctoPrint with my Mini and I admit, this is one of the functions, that I miss most. I would really like to be able to see remaining print time on the display.

Rose-David, 2022-01-18:
In my opinion, I'd rather that the Octopus logo only be used for when nothing's printing, but I can't see anything stopping the normal USB print UI from working here. Either way, I just wanna give my nice big LCD a nice big purpose.

neo666lis, 2022-02-04:
+1
This quite annoying not having the progress status.

coaxial, 2022-03-04:
As a reminder, #1497 is still open and failing. If someone who is more fluent in C than me could take a look, it would help us move closer to a fix.

pitops, 2022-04-18:
@coaxial did you check latest comment in the PR ?

coaxial, 2022-04-19:
> @coaxial did you check latest comment in the PR ?

Yes, didn't have the time to do it. Feel free to lend a hand.

pitops, 2022-04-19:
> > @coaxial did you check latest comment in the PR ?
> 
> @pitops Yes, didn't have the time to do it. Feel free to lend a hand.

Unfortunately i have zero C experience

coaxial, 2022-07-23:
I have rebased #1497 to `master` but I'm unable to do any further work on it. If anyone (or, who knows, Prusa themselves??) wants to carry on this work, knock yourself out!

Prusa-Support, 2022-08-02:
This is a quite hectic time for our developer team.

You may have noticed a rather large log for the recent [pre-release](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0-beta2
).
Probably you have also heard about other "big news" of announced products in the last year - which take up a little share of our developers' attention too.

Our developers will stick to a very tight and intense development plan for some time.

This doesn't mean your pull request will be forgotten. Your work is most valuable and truly appreciated, thanks.
Our devs will find time to dedicate to yours and the other pull requests as soon as possible. I hope you can understand.

Michele Moramarco
Prusa Research

lukkigi, 2023-01-15:
+1 interested in this as well

Prusa-Support, 2023-02-16:
No, I'm sorry, no news in this regard.
The news is that the ESP Wi-Fi module & Prusa Link implementation was successful and taken very positively by the customers.
Other than that, there is much more work going on under the hood - stay tuned.

The developers should find time to review this among a few more popular requests and I'm sorry I don't have an estimated time.

Michele Moramarco
Prusa Research

arekm, 2023-04-18:
About M73 reporting status. MK3* firmware supports that fine for years. MK4 firmware doesn't support this at all. So now this should be considered a functionality regression between MK3 and MK4 in i3 line.

Fortunately M73 support is already in Marlin2 (2.1.2+) and Firmware-Buddy is based on Marlin2. Unfortunately based on older version but backporting that code looks to be easy enough. It's guarded by M73_REPORT, SET_PROGRESS_PERCENT, SET_REMAINING_TIME and SET_INTERACTION_TIME ifdefs in the Marlin2 repository code.

Backporting that would get Firmware-Buddy into MK3 firmware feature parity regarding M73 g-code.

Relevant code is mostly:
https://github.com/MarlinFirmware/Marlin/blob/0c2ed8688dd387b56b318f6207ee1cc70811b5a5/Marlin/src/gcode/lcd/M73.cpp#L58-L91

plus ui.get/set_XX() functions plus a bit more if M73 would have to affect also what's displayed on LCD.

LordRaptor, 2023-04-18:
Please implement this, a lot of us are using Octoprint to control our printers and the standard set of G-Codes should be supported by all printers.

hstag, 2023-05-12:
Gentlemen, this situation is unacceptable. Despite purchasing the Prusa Mini printer three years ago, the feature in question has not yet been implemented. Furthermore, it is disappointing to note that even the latest model of the printer, the MK4, lacks this feature, which represents a downgrade from the previous MK3s model.

Prusa-Support, 2023-06-10:
Thanks for your solicitations, our developers and the whole company have great consideration for the community.
Our developers should stick to tight development plans at the moment but they will gladly look into this and other requests in the future. Until then, because the firmware is open source, I'm positive that the community's contribution can make a difference. 🙂  

For the records, the cameras are supported via Prusa Link (MK3 + Raspberry) or Prusa Connect.
Unfortunately, we still have a lack of documentation, but this was announced in the [release logs](https://github.com/prusa3d/Prusa-Link/releases), and here is a [list of supported cameras](https://github.com/prusa3d/Prusa-Link/blob/511dba9fd64a68f2362fe5355a67a024a7caede5/prusa/link/cameras/picamera_driver.py#L49). Alternatively, a "smart" camera device (smartphone, tablet, laptop...) can be connected via Prusa Connect very easily (QR code scan).

Michele Moramarco
Prusa Research

LordRaptor, 2023-06-20:
I completely agree with Bpendragon. Having the Mk4 regress in capabilities vs the Mk3 seems a very weird choice. And it's not like you guys don't have a firmware that does all of this, just allocate a developer that worked on the Mk3 firmware to port these capabilities to the Mk4, or review the pull requests from the community that integrate this functionality and merge them. You keep claiming that your developers have to stick to tight development plans, but this is an issue that's been open for over 3 years now, development plans don't go that far into the future.

fergymcferguson, 2023-06-22:
+1 Bump

oxivanisher, 2023-06-29:
Please work on this issue. Buying the MK4 feels like a big downgrade process and ease of use wise which is really sad!

Impatience71, 2023-06-29:
+1

I would also upvote this feature !

Soulofdestiny, 2023-07-04:
Yes please!

Jackjan4, 2023-07-04:
+1 Bump.

getgray, 2023-07-04:
Octoprint full compatibility is a need not a want.  Please implement ASAP

Jinra, 2023-07-17:
Having a 3+ year old issue for this feature doesn't bode well for the MK4. +1 anyway

TheNore, 2023-07-17:
Would it be possible to Please consider this a higher priority? As more mk4 units are being shipped more octoprint users are becoming frustrated and loosing functionality they have had in the MK3 printers for years!

ghoeffner, 2023-07-17:
Wait until you find out that [the filament runout sensor doesn't work with octoprint anymore](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3111)...

fergymcferguson, 2023-07-17:
Nor can you see any of the files on the USB drive...

Ikario, 2023-07-17:
This is a must. I get that Prusa is interested in people using their Prusaconnect/Prusalink option but it is nowhere close to have the same features as Octoprint.

jvasileff, 2023-07-17:
Pause print (`M601` added by PrusaSlicer) also doesn't work. I just built my MK4 yesterday, and I'm pretty disappointed that OctoPrint isn't really usable with it.

yahbluez, 2023-07-17:
i second that.

tideline3d, 2023-07-17:
24x MK3 farm here looking to expand soon. MK4 will have to be off the list for refusal to support 3rd party integrators with basic telemetry.  My entire production process is built around OctoPrint. 

I realize Prusa and Octoprint aren't on the best of terms but consider your customers rather than FOSS ecosystem politics here.

rnbarrett, 2023-07-17:
Count me in the upvote support!

osbock, 2023-07-18:
Upvote for me too! I just ordered the Mk4 upgrade, and had I known that Octoprint didn't work as well as it does on the Mk3, I would not have purchased.  I want to support you, please support us!

satchm0h, 2023-07-18:
+1

Please consider prioritizing support for the same interfaces/protocols/APIs the Mk2*->Mk3* supported. Your community have come to rely on many FOSS solutions that build upon these!

SalmonSays, 2023-07-18:
+1 for interest, this feels like a massive oversight in MK4 release.

shalmi, 2023-07-18:
@Prusa-Support PLEASE FIX THIS. It may not be important to you, but it IS important to your customers!

Monkopedia, 2023-07-18:
@Prusa-Support +1 to everything here. I back ordered an mk4 a few days ago, but now must consider cancelling it given the lack of support here.

[105 more replies]