prusa3d/Prusa-Firmware-Buddy #199: Problem unloading filament with Prusa Mini

gwildavies, 2020-02-11:
Just got a brand new Prusa Mini. Firmware 4.0.2+1977.
buddy board 1.0.0.

I'm having problems unloading and loading filament.
After a print, if I choose Unload filament, the extruder moves the filament backwards and forwards and then grinds the filament. I'm using the Prusament samples that came with the box.
The only way to remove the filament is to unscrew the brass fittings and pull the filament with pliers.
Could this be fixed, please? thanks!

Pitel, 2020-02-13:
Workaround: run "purge filament" before unloading. The tip of the filament is then soft and melted and passes all the tubes easily.

dtackgrounded, 2020-02-18:
> Workaround: run "purge filament" before unloading. The tip of the filament is then soft and melted and passes all the tubes easily.

The issue described is not related to a blob forming at the tip of the filament. The extruder sometimes grinds the filament on unload operation until it is unable to 'bite' into the filament anymore. Once the extruder gear has lost grip on the filament, the only way to remove the segment is to do what OP described.

bildzeitung, 2020-02-23:
I’ve found this as well. Did a purge, then unload, and had to take apart the tubing to extract the pieces of filament that had been ground apart.

michalxfanta, 2020-02-24:
In 4.0.3 firmware we improved unload sequence.

closed 2020-02-24 by michalxfanta

bildzeitung, 2020-02-24:
I’m running the 4.0.3 firmware and the last two (only two! brand new mini!) filament changes chewed up the filament & required disassembly to fix.

I don’t think this is fixed in the new version.

augied, 2020-02-24:
I also have this problem with 4.0.3.

When trying to unload, the tip of the filament is too large and jams in the tube.  Once it's stuck, the extruder starts grinding the filament.

bildzeitung, 2020-02-24:
Further to this — I tried running a purge before the unload and it was still filament snack time :(

JakubKohoutek, 2020-03-01:
In my case there is also still a blob at the end of filament (even with a 4.0.3 firmware) that gets stuck and cause extruder to chew the string. IMHO this shouldn't be closed.

simonbugler, 2020-03-01:
I believe this is a hardware problem. I had it too and did step 9 of this page (reseating the hot end) solved my problems. 

https://manual.prusa3d.com/Guide/How+to+change+a+hotend+PTFE+tube+-+MINI/1505?lang=en#_ga=2.105683011.1752830996.1583083336-1537360900.1583083336

BorisDess, 2020-04-29:
Tip was properly cut but blocked in the extruder despite asking 3 times for extraction. I got to pull hard to get it out. Meh.

leehambley, 2020-04-29:
I can "confirm" this is a hardware problem, lots of issues with the PTFE tube are reported in the Prusa forums, a proper by-the-book rebuild as linked by @simonbugler is the solution I needed.

Word of warning, pre-heat your hotend, use gloves, and don't try and remove the brass nozzle when cold *snap*.

isopterix, 2020-06-13:
Yeah :( Just ended up having the same problem here after some great printing. First time I wanted to use a different color... and... boom.

bildzeitung, 2020-06-13:
I haven’t seen this issue in v4.0.5, but I also changed the hotend PFTE tube & moved up the hotend a millimeter.

I’m currently thinking that unload problems are a symptom of the hotend + PFTE issue, & that the unload sequence is now ok.

x10102, 2020-06-14:
I also had to rebuild the hotend before I could even print. But maybe the filament grinding issue could be caused by drive gear tension? (Too tight or too loose)

bildzeitung, 2020-06-14:
Yes, the idler screw definitely plays into it; I don’t have the best sense on that myself, yet, but so far it seems to be working.