prusa3d/Prusa-Firmware-Buddy #200: Feature request while loading filament

gwildavies, 2020-02-11:
Brand new Prusa Mini, firmware 4.0.0. Using the sample Prusament that came with the printer.
whilst loading filament, often the filament doesn't get immediately pulled into the extruder, so when the extruder stops, the filament is only half way along the bowden tube.

If 'is color correct' = No, then
Filament moves a bit further forward then 
We're sent back to main menu. At this moment I have a half-loaded filament only.
Add loop back to No, to give an option to extrude a bit more and eventually purge? 
Thanks!

x10102, 2020-02-18:
You need to push the filament in until the extruder catches onto it. You can also manually move the extruder from the settings > move axis menu

simonbugler, 2020-02-22:
I have to manually load the filament the first cm (until it 'catches') using my thumb to turn the extruder cog. 

The automated 'Load Filament' sequences needs some work as too often it continues when it's not successfully 'caught'. and so you end up with half loaded filament.

I would like to see during the 'Load Filament' sequence the option to turn the dial to adjust the extruder, to save using my thumb on the cog.

michalxfanta, 2020-02-24:
Purge function is implemented in the latest firmware for MINI.

closed 2020-02-24 by michalxfanta