prusa3d/Prusa-Firmware-Buddy #212: DHCP broken on 4.0.3-RC1

koenkooi, 2020-02-13:
With 4.0.1 the mini would do a DHCP discover and then use the IP it was handed by the server, but on 4.0.3-RC1 it doesn't take the IP:

```
Feb 13 10:26:48 apu2c4 dnsmasq-dhcp[30300]: DHCPDISCOVER(enp3s0) 10:9c:70:08:0a:fc
Feb 13 10:26:48 apu2c4 dnsmasq-dhcp[30300]: DHCPOFFER(enp3s0) 172.20.0.168 10:9c:70:08:0a:fc
Feb 13 10:26:56 apu2c4 dnsmasq-dhcp[30300]: DHCPDISCOVER(enp3s0) 10:9c:70:08:0a:fc
Feb 13 10:26:56 apu2c4 dnsmasq-dhcp[30300]: DHCPOFFER(enp3s0) 172.20.0.168 10:9c:70:08:0a:fc
Feb 13 10:27:12 apu2c4 dnsmasq-dhcp[30300]: DHCPDISCOVER(enp3s0) 10:9c:70:08:0a:fc
Feb 13 10:27:12 apu2c4 dnsmasq-dhcp[30300]: DHCPOFFER(enp3s0) 172.20.0.168 10:9c:70:08:0a:fc
Feb 13 10:27:44 apu2c4 dnsmasq-dhcp[30300]: DHCPDISCOVER(enp3s0) 10:9c:70:08:0a:fc
Feb 13 10:27:44 apu2c4 dnsmasq-dhcp[30300]: DHCPOFFER(enp3s0) 172.20.0.168 10:9c:70:08:0a:fc
Feb 13 10:28:44 apu2c4 dnsmasq-dhcp[30300]: DHCPDISCOVER(enp3s0) 10:9c:70:08:0a:fc
Feb 13 10:28:44 apu2c4 dnsmasq-dhcp[30300]: DHCPOFFER(enp3s0) 172.20.0.168 10:9c:70:08:0a:fc
Feb 13 10:29:44 apu2c4 dnsmasq-dhcp[30300]: DHCPDISCOVER(enp3s0) 10:9c:70:08:0a:fc
Feb 13 10:29:44 apu2c4 dnsmasq-dhcp[30300]: DHCPOFFER(enp3s0) 172.20.0.168 10:9c:70:08:0a:fc
```

[image]

jnoxon, 2020-03-07:
On 4.0.4 I'm not even seeing a discover.

Lino77, 2020-03-08:
I can confirm

koenkooi, 2020-03-08:
The [4.0.3 release notes](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.0.3) are very clear on this:

> Ethernet instability
In some scenarios, the Ethernet connection of the MINI 3D printer can be overloaded with requests, which may lead to a system crash. The fix will be available in the next version of the firmware along with a new web interface. In the meantime, we have decided to block the Ethernet networking service in order to ensure a safer experience.

So:

4.0.2: broken DHCP
4.0.3: No ethernet at all
4.0.4: No ethernet at tall

gjsman, 2020-03-20:
Works on 4.0.5-RC1, thus resolved.

closed 2020-03-20 by koenkooi