prusa3d/Prusa-Firmware-Buddy #220: Lower err_mintemp threshold?

x10102, 2020-02-15:
So I like my room cold, and the nozzle and heatbed temperature often drops to around 15°C.
This triggers the err_mintemp failsafe and I have to heat up my room or warm up the nozzle with a lighter. Maybe it would be a good idea to lower the err_mintemp threshold from 20°C to 10°C? With an incorrectly wired thermistor the temperature is gonna be either 0°C or extremely high anyway.

petrzelka, 2020-02-17:
Lowering to 10C would be great. I'm using hairdryer to heat up to >15C (from 14C).
Also, since update to latest RC, it doesn't show mintemp error, but it resets and after boot shows watchdog reset red screen.

loglow, 2020-02-20:
I'm going to add my support for this change as well. When I first setup the printer I got this error immediately and lost a decent amount of time trying to figure out what was going on (reading online, checking all the cables, etc.) just to discover that the room was a few degrees too cold. My printer is in a space that's regularly in the 50-60 degrees (F) range, especially during the winter.

masto, 2020-02-21:
This is a pretty big issue. Especially since the 4.0.3 firmware doesn’t display a MINTEMP error, it just watchdog reboots. I spent a bunch of time after assembling my brand new printer trying to figure out why I couldn’t enter the setup wizard or start calibration without it crashing. Only after downgrading the firmware, getting the err_mintemp message, and searching around on Github, did I actually discover the workaround.

skiselev, 2020-03-28:
Same here. The room temperature is 15C. The printer refuses to do anything unless I turn up the heat in the whole house. Not particularly energy efficient.

masto, 2020-03-28:
> Same here. The room temperature is 15C. The printer refuses to do anything unless I turn up the heat in the whole house. Not particularly energy efficient.

I find it's enough to just hold the print head with my fingers for 10 seconds or so, until it gets above 15C, and then I can start printing. I still think the software could compensate for this situation better, but I find this to be the easiest workaround.

JohnnyDeer, 2020-12-14:
Hi, in version 4.3.0-RC1 the mintemp is lowered to 10 ° C. I close this issue as solved.