prusa3d/Prusa-Firmware-Buddy #221: Heaters stay on after loading filament 4.03 RC

simonbugler, 2020-02-16:
The heaters are staying on when loading a new filament.

x10102, 2020-02-17:
I think it's intended. When you're loading a new filament you're probably going to print soon.

simonbugler, 2020-02-22:
When I load filament I'm loading filament.
When I'm preheating I'm preheating.
When I'm printing I'm printing.

I don't really want the interface to assume I'm going to do anything in the future.

Vic-Y, 2020-04-11:
Since there is a "Change Filament" menu item (where it makes sense to leave heaters on between unloading and loading), "Load Filament" should turn heaters off after completing.

x10102, 2020-04-11:
Why would you want to load filament without doing anything else?

Vic-Y, 2020-04-11:
I have one use case where this occurs, but it may not apply to many other people. At work, our printers are located on a different level of the building from our design desks. We monitor and start jobs remotely via OctoPrint. When a print is finished and we remove it from the printer, we commonly change spools to prepare for the next print if the printer is low on filament. However, the designer may not be ready to start the next print for several hours, so it would be better for the heaters to turn off instead of staying on.

JohnnyDeer, 2020-05-13:
After implementing safety timer to all situation when printer stay on temperatures, will trigger cooldown after 30 mins. Right now safety timer work in preheat only. Seting cool down automaticly after load is to internal discusion, as well as suggestion in #439 .

Prusa-Support, 2022-09-23:
Thank you all for participating in this discussion and sharing your ideas.

It seems that this issue was fixed by the implementation of a _safety timer_.
Moreover, this issue seems stale at this point.

Please allow me to close this issue, and please mind that you can find a _cooldown_  option available in various [menus](https://help.prusa3d.com/article/lcd-menu-mini-mini_207531), or simply press the **reset button** next to the knob, in order to interrupt the heating right away (no need to wait for the _safety timer_ counter).

Michele Moramarco
Prusa Research

closed 2022-09-23 by Prusa-Support

ThomasCZ, 2022-09-23:
It is not right, because heat bed is also heating and whole process is waiting for that. When I am unloading a filament, it is just wasting of energy. I think, it is wasting of energy in 100% of cases, because when I select "unload filament", I want to unload a filament. If I wanted to print with another filament, I can select "Change filament". Safety timer does not solve wasting of energy.

Anyway, could you please add at least some "eco" check box, that will not heating heat bed and turn off heating immediately?

Prusa-Support, 2022-09-27:
Sounds good to me, but kindely mind we have a separate issue about that.
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/706

Michele Moramarco
Prusa Research