prusa3d/Prusa-Firmware-Buddy #230: Beep buzzer after print is finished.

x10102, 2020-02-17:
After finishing a print and homing the printer could beep a few times to let the user know the print is finished.

loglow, 2020-02-20:
I second this request, but please make it an optional setting.

x10102, 2020-02-21:
> I second this request, but please make it an optional setting.

Yes, it could be enabled by default and you could disable it in settings as it can be annoying for some

michalxfanta, 2020-02-24:
Thank you for an interesting idea, we will consider its implementation.

om2kw, 2020-02-24:
beep at the end can be done byM300 gcode (play tone) as a last command.
M300 should be supported according to marlin docs

x10102, 2020-03-01:
> beep at the end can be done byM300 gcode (play tone) as a last command.
> M300 should be supported according to marlin docs

M300 doesn't do anything

JohnnyDeer, 2020-05-21:
Updated informations, in next FW version will be sound notifications. M300 will still not being supported.

closed 2020-06-14 by x10102