prusa3d/Prusa-Firmware-Buddy #241: Interface flashes during bed probing/leveling (minor issue) 4.0.2

loglow, 2020-02-20:
During bed probing/leveling the interface displays a message about which point is currently being probed (eg. 8/16, 9/16, etc.) but the message doesn't remain on-screen for long enough to make it to the next probe point, so as a result the interface flashes quickly away from the message, and then flashes right back to the next probe message. This is a minor issue, but it appears sloppy.

While probing the bed the status message should remain on-screen continuously and update the current probe point as it moves through them, without flashing back and forth between screens.

I haven't updated to 4.0.3 yet so I don't know if this issue has been addressed, but I didn't see it mentioned in the release notes.

gjsman, 2020-03-01:
Fixed in 4.0.3, so I'd recommend closing. (I have personally witnessed this fixed.) Also, it is mentioned in the Release Notes:

Optimized on-screen messages
The management system responsible for displaying the messages on the printer's screen was updated to prevent overlapping messages, which sometimes caused the screen to flicker.

x10102, 2020-03-01:
This is fixed in 4.0.3

loglow, 2020-03-02:
I've updated my printer to 4.0.3 now and can also confirm that this was fixed.

closed 2020-03-02 by loglow