prusa3d/Prusa-Firmware-Buddy #242: Move tool higher after first layer calibration wizard

Crawlerin-SA, 2020-02-22:
After first layer calibration wizard finishes, it parks at home position right above surface. It would be good if tool gets raised higher, to be able to remove and clean steel sheet more easily.

jurcja1, 2020-02-27:
For me its better to park at Z0,4, because of possible filament leak during probe heating to print temeperature. Simply changed start g-code in slicer.

Area5142, 2020-02-28:
Related to closed issue: _Feature Request: Lift and Park after "First Layer Cal."_ #50

loglow, 2020-03-02:
This is a good suggestion.

In addition:
- The build plate should be extended all the way forward after the first layer calibration is complete, as it is after a regular print finishes.
- During a regular print the bed probing is performed at 170 degrees, but during first layer calibration the bed probing is performed at the final (much higher) nozzle temperature.
- The first layer calibration pattern (in 4.0.3) leaves a small gap between the last horizontal extrusion and the perimeter of the rectangle that's being filled in.

JohnnyDeer, 2020-12-14:
Hi, in 4.3.0-RC1 is FLC procedure changed and that solves this request. I close this issue as solved.

closed 2020-12-14 by JohnnyDeer