prusa3d/Prusa-Firmware-Buddy #243: Unable to print files with very long names

bildzeitung, 2020-02-23:
If the file name is long enough, it shows up in the UI as the Windows short form (eg FOO~1.GCO).

When the file is selected, it’s rejected (can’t see the error, if there is one, it flashes by too quickly) & the UI returns to the main menu.

However, if I rename the file to something shorter, all is well (so definitely not a corrupted GCode).

It’s unexpected, so I thought I’d mention it.

jurcja1, 2020-02-27:
Similar to this behaviour is error when gcode is not in \root\folder but \root\folder\folder. Then when you initiate gcode, it shows info, but then is time 0, rest 0 and printer does nothing.

mlgold, 2020-04-28:
Not sure if this is the same issue or not:

https://forum.prusaprinters.org/forum/hardware-firmware-and-software-help/long-filenames-dont-show-on-print-list-in-fw-4-0-5/

Long filenames remain an issue,

bildzeitung, 2020-05-07:
> Not sure if this is the same issue or not:
> 
> https://forum.prusaprinters.org/forum/hardware-firmware-and-software-help/long-filenames-dont-show-on-print-list-in-fw-4-0-5/
> 
> Long filenames remain an issue,

Sounds like it’s the same one; glad it’s not just me :)

michaelaswood1980, 2021-03-28:
I think I have experienced the same issue today. I have several GCODE files in a folder structure several levels deep and when I came to print it today (for the first time) I got a blue screen as shown below for about 15 seconds before crashing and displaying the orange error code about the hotend temperature and check your wiring.  After you restart the printer I also noticed that the loaded filament type you have in the printer resets to the default PLA one.  I don't know if this is related to the Blue screen error or if this is done by design...
This is while using the latest BETA Firmware 4.3.2.
[image]

zoltan-l, 2022-01-20:
the recent FW 4.34 allows at least the file name of 90 characters, which should be more than sufficient.

bildzeitung, 2022-01-20:
IIRC, there is now a completely different file system library.

I have a feeling this issue is no longer relevant, as you imply, and can probably be closed.

I certainly haven't experienced it in recent firmware versions.

zoltan-l, 2022-01-20:
Thanks for response , I will close it

zoltan-l, 2022-01-20:
Not relevant enymore

closed 2022-01-20 by zoltan-l