prusa3d/Prusa-Firmware-Buddy #249: Add long filename scrolling

julianrupp, 2020-02-24:
See title. Feature analogous to how it's currently implemented in the i3 firmware - if filenames are too long to display, make them scroll across the screen slowly - in the file selection menu as well as on the single file preview and printing screens.

I embed a lot of information in my gcode file names for better overview before I even select a file to inspect. The current interface is pretty unusable to me, as even the model name is too long most of the time to push the filament type tag which PrusaSlicer puts in the default file name off the screen.

Area5142, 2020-02-28:
I opened the issue in December here: "Only first 19 letters of long filename is shown" #25

In the closing statement it was promised to be in this FW version...

I too have long filenames with a lot of info, but it is hard to tell them apart when only the first part of the filename is shown.

gjsman, 2020-03-01:
It appears Prusa was really underestimating the launch times for their software, as you can read their PrusaPrinters.org and they thought the web UI would be finished and publicly available in late January / early February, which didn't happen. Instead, they've had the bug fixes 4.0.2 and 4.0.3, which don't come off as exactly _planned_ (i.e. they are a bit behind their schedule). I wouldn't be surprised if, when 4.1 or whatever it is called is released, that it appears then.

loglow, 2020-03-02:
Underestimating launch times seems to be human nature. I find it both extremely frustrating and completely relatable at the same time.

freshmas, 2020-03-17:
With the additional resolution of the mini display, I'd rather see the long names wrapped instead of scrolling.

This will probably be a non-issue once the web interface is available, but still.

spaxi82, 2020-04-17:
I would also be very happy about this feature. Unfortunately I can only see the first part of the file names and therefore all files look the same.

rmeister345, 2020-07-02:
Please fix this.
  I too like to name my files and include the file version or rev, the material, noz dia, noz temp, bed temp, what printer it's for, the date and even how much filament is required for the print.  Some of the time it works and then some of the time my names are too long and I have to start cutting up my file names which make this is a really annoying issue for such quality printers.
No worries, we all have little quarks and priorities and **_I do appreciate the new products and the personal touch you guys bring to it_**, just don't forget the little things.  @MRprusa3d

hawwwran, 2020-07-02:
> Please fix this.
> I too like to name my files and include the file version or rev, the material, noz dia, noz temp, bed temp, what printer it's for, the date and even how much filament is required [...]

It was already fixed, Will come in new version soon.

julianrupp, 2020-07-02:
Already saw it in release notes for 4.1.0-RC1 and waiting for the full release :)
A lot of stuff in there that finally should bring the Mini firmware very close to the level of the i3 one in terms of QoL. Should have been there from the beginning, but better late than never.

freshmas, 2020-07-02:
I get your point, but frankly I’m glad to have had the opportunity to use the mini in the state it was released. When can I get to try the XL? Haha

JohnnyDeer, 2020-07-08:
Hi, in FW version 4.1.0 is long filename scrolling implemented in file selection menu. Please try it out :)

julianrupp, 2020-07-08:
Already saw the release candidate, but waited for the final release - I won't install anything else on my machine. A much needed improvement! Would love to see a user setting for scrolling speed in a future version though - it's a bit slow.

gjsman, 2020-07-08:
@julianrupp I have been printing with every RC so far. You might be surprised, as the RCs have generally been very stable. Obviously, don't trust a RC for total stability and this is the mandatory "I haven't been burned yet!" message, but I think they are generally pretty good. This is also indicated by that there is only one RC per release and it is almost an actual "Release Candidate," not five beta releases like some companies.

Vic-Y, 2020-08-09:
Hi. Long filename scrolling is working in 4.1.0 and now also 4.2.0. However, scrolling speed seems slow. Please consider scrolling a little faster.

JohnnyDeer, 2021-02-16:
File name scrolling is already implemented for every screen, I´m closing this issue.

closed 2021-02-16 by JohnnyDeer