prusa3d/Prusa-Firmware-Buddy #255: Watchdog reset

Pitel, 2020-02-27:
After whole day and night of printing, the MINI stopped mid-print, beeped and displayed red error message WATCHDOG RESET. :scream_cat:

Latest firmware 4.0.3

There is a crash dump: [dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/4264087/dump.zip)

Pitel, 2020-02-27:
After that, it prints again.

Pitel, 2020-03-28:
Happened again, mid-print. FW 4.0.5 RC1

[dump2.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/4397220/dump2.zip)

Pitel, 2020-04-02:
It happended again yesterday and today. Between that, I managed to print the part fine twice (using _reprint_ function), so it's not corrupted gcode. Unfortunately, my wife was there, so I couldn't get the error dump this time. No specific error on the screen, just _press to continue_. Still FW 4.0.5 RC 1.

Pitel, 2020-04-05:
Happened again. And again, 1 hour before end of 16 hour print. FFFFFFFFFUUUUUUUU...... :rage4:

All the crash `dump.bin`s have the same md5 as ythe one in `dump2.zip`. So I won't upload it.

I suppose it should be fixed ASAP, because pople are also complaining about it [at the forums](https://forum.prusaprinters.org/forum/hardware-firmware-and-software-help/prusa-mini-resets-itself-showing-watchdog-reset/).

Pitel, 2020-04-06:
@michalrudolf @michalxfanta @RadekVana @XPila @DRracer Could some of you guys help with that? What does the dump says?

DRracer, 2020-04-06:
@Pitel in the next FW we'll solve the USB disconnecting problem while printing. There is one option you may try - do you have a USB extension cable? Can you please try to plug the USB flash via the extension cable into the MINI and report back if it makes any difference?

Pitel, 2020-04-06:
@DRracer Thanks for reposnse. :+1: So the cause is USB disconnecting during print? :scream_cat:

Actualy, I have short (30 cm or so) 90 degree USB extension cable which I keep using for easier access to USB (I have my MINI next to a wall on the right side). Should I try removing it?

DRracer, 2020-04-06:
@Pitel it might be judging from your description. Please try to remove the cable and stick the USB flash directly into the MINI if anything changes.

Pitel, 2020-04-06:
I thought you could get that information from the crash dumps. But ok, I'll try it with the extension and report back after couple of days.

Pitel, 2020-04-07:
I upgraded to 4.0.5 and tried to start some simple print and remove the flash dfrive manualy after a while (still using the extension cable). The print paused and hotend moved away. If this is the change made in 4.0.5, than good job. :+1:

But... after inserting the drive back and pressing resume, the print didn't resumed and the menu/screen was frozen so I had to reboot anyway. :-1:

There was also a bit weird white border around the pause/resume button after removing the drive.

Pitel, 2020-04-07:
@DRracer Got it again with 4.0.5. So I guess it's not USB disconnect. :(

I'll provide `crash.dump` later today.

Pitel, 2020-04-07:
[dump5.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/4444726/dump5.zip)

hansamann, 2020-04-11:
I got my Mini today and got this error half way into the first print. I was then lucky and my short ~50min print succeeded - the prusa logo on the supplied usb stick. I got more adventurous and tried a 1:20 print and got this error half way into the print. 

This is super annoying....

hansamann, 2020-04-13:
I tried preheating to PLA and wiggled the cables as support told me - no error occured. I started printing and about 1h into the print the error occured again. I cannot use the printer this way at all...

Pitel, 2020-04-14:
Ok, it might be USB related. Two things happened in fast succession: I've removed the extension cable and USB filesystem got severely corrupted and I had to reformat it. I don't think they are related, me or my wife might have pulled the stick from comupet too soon. Anyway, I've not seen the error in a couple fo days.

I'll try to put the extension cable back and see if it changes.

I migth also try running `badblocks` on the drive and see if it corrupted.

regularfry, 2020-04-16:
I hit this on 4.0.5, after a successful Benchy print.  I've got no extension cable, but it's not impossible that the drive might be vibrating in the socket.  I'll try holding it in place with some blue tack and see if it goes away.

yck011522, 2020-04-16:
Allow me to joint the party. I am getting this mid-print error too. Super frustrating. 

- [x] Cable wiggle not revealing anything.
- [x] Crash on file given in USB from factory / sliced it myself with PrusaSlicer
- [x] Crash at random location.
- [x] Tried the stock USB stick, another stick, same thing. (Direct plugin)

Are developers able to reproduce the error?

Darkwing69, 2020-04-16:
Same issues as above. I have tried everything and 10 different stl files, usb sticks, direct connect and keeps failing. 

This is brand new 2 days old. Maybe I should have got a ender.

Barbouri, 2020-04-16:
Running 4.0.5 and had three separate watchdog resets. I tried a new USB flash drive and the issue went away, until I plugged the silver Prusa flash drive back in, and failed 5 min. into the print.
Suspect that the Prusa flash drive is faulty after printing for a whole day with no issues.

Darkwing69, 2020-04-16:
> Running 4.0.5 and had three separate watchdog resets. I tried a new USB flash drive and the issue went away, until I plugged the silver Prusa flash drive back in, and failed 5 min. into the print.
> Suspect that the Prusa flash drive is faulty after printing for a whole day with no issues.

I'll give another flash drive a try.

regularfry, 2020-04-17:
Bear in mind that if the machine resets, the USB drive could potentially end up corrupt just because FAT32 is rubbish.  That could well be a symptom, not the cause.

yck011522, 2020-04-20:
Okay, I managed to switch to a different USB stick. 
**SanDisk Ultra Dual 32GB**
And it worked find for two 10+ hours prints in a row.

However, I run into another problem that may or may not be related.
https://photos.app.goo.gl/b9sZYHfWFQACSBvM9

regularfry, 2020-04-20:
I got a customer support response saying that (in my case) it's probably bad gcode being put out by PrusaSlicer, and to try reslicing with the latest version.  I already had the latest version, but I resliced anyway at the `20% SPEED` setting.  *That worked*, and I've had a few more successful prints since then. Given that the faster print head movement shakes the printer more, makes me think it's not a mechanical issue for me.

However, it would mean that there's a problem with the combination of the current latest MINI firmware and PrusaSlicer gcode generation at (at least) the `15% QUALITY` setting.

Pitel, 2020-04-22:
Alright, my flash drive (the one which came with the printer) failed again. So I ran `badblocks` on it, and it reported almost 25000 errors. So I guess it's done.

I'm considering closing this issues, because it was obviousy not a firmware bug. But there are other people, some with similar symptoms, some with different ones, so I don't know... I'll let the devs decide.

Pitel, 2020-04-24:
Don't close. I got it again yesterday, with news USB stick. But with extension cable.

polarvogel, 2020-04-26:
This is my second Watchdog Reset mid-print on my new (~1 week) Prusa Mini:
- Mini Firmware: 4.0.5; PrusaSlicer 2.2.0
- both Watchdog Resets happened mid-print on PrusaSlicer gcode
[dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/4535042/dump.zip)

krcm0209, 2020-04-28:
I have just experienced my first watchdog reset. Right after it happened I tried to save a dump to the flash drive and it failed saying that no drive was found even though the drive was plugged in. Anyway, I unplugged and replugged the drive and here is the dump: 
[dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/4548991/dump.zip)

krcm0209, 2020-04-29:
Just got my second watchdog reset in 24 hours. This is getting ridiculous. Never once did I get an error like this on my old MK2. 

Dump: [dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/4553248/dump.zip)

Pic of first print where it reset: 
[image]

Pics of second print where it reset:
[image]
[image]

Bertware, 2020-05-01:
I got a watchdog reset 3 hours into a 9 hour print, using the factory provided USB drive, with a model sliced using PrusaSlicer 2.2.0.

The print and profile can be found here: 
https://www.prusaprinters.org/prints/21794-tolerance-gauge-fidget-toy/files

I moved it to the center of the heatbed, and used my own filament profile, but that's all.

[prusa mid-print watchdog crash.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/4564074/prusa.mid-print.watchdog.crash.zip)

krcm0209, 2020-05-01:
I was also using the USB drive supplied by Prusa.

Bertware, 2020-05-01:
Happened again on the same model (after re-slicing), at a similar layer (might be a few layers of, but within +/- one millimetre from each other).

Reading the gcode from the USB stick was impossible, but I included the gcode from my computer (an identical copy)

[prusa crash mid print.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/4564718/prusa.crash.mid.print.zip)

hansamann, 2020-05-02:
Since switching to another USB drive, I did not have the problem again - I am using the latest 405 firmware and gcode that I run myself through prusaslicer. This would be my recommendation to all: dump the Prusa supplied USB stick.

yck011522, 2020-05-04:
@hansamann I agree with you.

On another note. I suggest Prusa investigate into the root of this issue. Since this is not just 1 or 2 USB stick at fault, it could be some types of USB drive are simply not compatible.

krcm0209, 2020-05-06:
I can confirm so far that switching to a different USB stick from the one supplied by Prusa with the printer has resolved my watchdog reset issues.

Bertware, 2020-05-06:
I can also confirm using another USB drive solved the issue. The provided USB stick is probably of poor quality. However, the firmware should

- Notify if the USB stick is at fault
- Allow to resume a print after rebooting

benoitcosials, 2020-05-11:
I had the same problem. From the delivery, when I tried to print the prusa logo or the boat, I got a watchdog reset every time at the same place. To solve it, I reformat the USB key with the macOS secure erase (it wright data x times) in order to verify if I got physical error. The format went well, I copied back the model without reslicing them. Every print worked well after that...

From my point of view, it seems there is something wrong with the USB key, but not sure if it's physical or just a bad formatting / corruption. Hope that help.

regularfry, 2020-05-11:
Just going to +1 the "switching to a different USB stick solved my problem".  Seems to only be a problem on files above a certain size, but that might just be because there's something random going on and a longer file gives it more of a chance to happen.

heitzer95, 2020-05-22:
I have the same topic with my Mini, delivered today. Really annoying. 

Gonna try another usb-drive tomorrow.

kdpainter, 2020-05-26:
I just started getting watchdog resets after a couple of months without seeing a single one.  Now it is resetting every single time when it is preparing to do the mesh level.  It doesn't even make it to the mesh level now.  I have tried several different thumb drives.  Same result.    I am on v4.0.4 due to another problem that popped up with v4.0.5.  I do not have a USB extension cable.  I may try reflashing the same version of firmware and doing a factory reset.

jbur4556, 2020-05-27:
I have been having the same issues with my Mini that was delivered today, trying a new USB stick, will update if the Benchy finishes or not.

kdpainter, 2020-05-27:
So far, this seems to me to be bugs with PrusaSlicer.  There was a configuration update today.  Today is when I started seeing this problem.  I also use Simplify3D and have a profile that works reasonably well with PETG and the Mini.  I tried it to print a Benchy and it is printing.  What is coming out of PrusaSlicer causes a watchdog reset immediately.  Perhaps this config. is intended for v4.0.5?

PrusaSlicer seems to produce better quality prints than Simplify3D which really shocked me but there is something wrong right now.  Hopefully it will get fixed.

UltraMej, 2020-05-29:
I just received my MINI yesterday, and got the same "Wachdog reset" issue here:

-1 Printed the frog that was already on the Prusa USB stick: printed ok.
-2 Printed a "big" model i sliced myself (prusaslicer 2.2.0) onto the Prusa USB stick: Watchdog error at ~10% print
-3 Re-sliced my big model on another computer onto Prusa USB stick: Watchdog error at ~5% print
-4 Printed the nut that was already on the Prusa USB stick: Watchdog error at ~8% print
-5 Printed the *same Gcode as -3* but from my own USB stick that I quick formated (FAT32) just to be sure: No Watchdog error.

For me: The 3€ Prusa USB stick is to blame. Not the Gcode. Not the Slicer.

kghunt, 2020-06-03:
I am not sure this is a firmware bug. I had 2 out of 4 of my first prints fail with this error. After switching to an 8gb Kingston USB drive I had on hand the issue has not re-occured 10 prints later. It seems to me the supplied USB drive is of poor quality.

scubagregory, 2020-06-09:
I recently experienced this with one of my minis.  After some research on the forums, I narrowed it down to the USB stick sent with the printer.  I tried formatting the USB stick.  It loaded the files from my PC fine but had the watchdog timer error at the second layer.  I reset and used a name brand USB drive.  No issues since swapping out the USB stick.  I contacted Prusa support and they are sending me another USB stick.  However, I was really wanting Prusa to check their batch of USB sticks to see if they are too slow or failing or what.

kghunt, 2020-06-09:
It was definitely the USB for me. 2 out of my first 4 prints failed. Since switching to the Kingston not a single failure in dozens of prints.

TJ-G, 2020-06-15:
I'll confirm that switching out the USB stick solved the issue with prints on my mini as well.

JohnnyDeer, 2020-06-22:
Hi @Pitel , this issue is solved in 4.1.0-RC1. Please try it out :)

kghunt, 2020-06-22:
Update: probably 100 hours printing and no failures since switching to a branded USB stick.

JohnnyDeer, 2020-07-21:
No reaction, I´m closing as solved.

closed 2020-07-21 by JohnnyDeer

jfmmm, 2020-10-09:
Also got a crappy usb key with my mini as of September 2020.. Printing the nut failled at around 8% twice in a row. Switched to a personal USB key and it solved the issue.

roborebels, 2020-10-23:
I also have run into the failing USB drives.  Out of the 5 Minis that I have 2 USB drives have failed so far.  One of my printers isn't accepting any USB drive that I try.

JohnnyDeer, 2020-11-03:
Hi @roborebels please contact our tech support. Faulty USB is not related to FW. Thank you.