prusa3d/Prusa-Firmware-Buddy #257: Crash on print stop

Bjerg, 2020-02-28:
problem is that it crashes into x or y homing position sometimes both depending on where it is while stopping print because it seems like it move a fixed about and not to the specific position compared to where it is.

xavor92, 2020-02-29:
Can confirm, same issue on mine

jurcja1, 2020-03-01:
I confirm that. Had same issue several times depending on variable stop position, x axis crash and then chaotic behavior when doing manual move of axis through menu.

x10102, 2020-03-01:
Probably related to / a duplicate of #27 and #16

keksa, 2020-03-02:
I had similar problem when moving the axes manually from settings menu (right **after** auto home calibration). The printer would try to go over where it could and I got about a 2 second screech before it determined it really can't go there.

jhenley01, 2020-03-05:
Same here, I recorded a video of it happening.

[https://youtu.be/HSwy_sHWCMo](https://youtu.be/HSwy_sHWCMo)

HenrikPankoke, 2020-03-09:
I had a print with 6 elements printing individual.
The elements is place like the photo. print is from back left to back right.
moving to front left to front right

[image]

When printing the front right element I stopped the print.
the printhead then moves backwards an into the element behind. causing this to be knocked of the plate. In this case it was a small part and not theat high adhesion. But larger parts with could adhesion could have damage that fan on the printhead.
[image]

maybe changing stop command to move the nozzle up above highest printed element, before moving in any other direction. or just move nozzle clear of current print and stop movement.

jurcja1, 2020-04-09:
In new FW 4.0.5 is different behaviour of "Stop print", I think it is also faulty.
When you stop print, print head stops at same place of last movement in printed object without any movement out of object.

JohnnyDeer, 2020-06-30:
Hi, this bug should be repaired on 4.1.0-RC1, please try it out. Thank you.

JohnnyDeer, 2020-07-21:
No reaction, I´m closing as solved.

closed 2020-07-21 by JohnnyDeer