prusa3d/Prusa-Firmware-Buddy #260: load filament UX

keksa, 2020-02-28:
When loading filament for the first time ever, I did not push it all the way to the gears in time. The printer did not recognize this, continued the process as if it had the filament loaded and just asked me if the color was correct. 

I was confused and clicked the "no" option to indicate the color was not correct - since there was no filament being extruded, but it just started purging (with nothing extruded) and did not try to grab the filament again.

I find the UX weird - "yes" means everything is fine, but in this case it's not; "no" means the color is incorrect, but there is no color. It might be nice if there was a 3rd option to just try loading the filament again.

loglow, 2020-03-02:
I had the **exact** same experience when I loaded filament for the first time.

MichaelReitsma, 2020-03-03:
it would also benefit from  another option such as load more . I have also had the case that it grabbed the filament but very late in the process. Purging would be the only way to get it to advance the filment. However that is very slow. So exit the menu and do a load filament again , which seems counterintuitive.

julianrupp, 2020-03-20:
I also noticed there is no beep to inform you that pre-heating has completed if you choose to load filament when the printer is cooled down. Since heating up takes a while, one is tempted to start the process, then walk away for a bit. A beep would inform you in a timely manner that the printer is ready for you to continue.

Closely related: Unloading filament UX. No beep here either if the printer is not up to temperature yet when starting the process. What's worse, the printer will automatically continue to eject the filament instead of waiting for you to press a button, which means it'll screw up your filament spool's winding if you walk away while the printer heats up.

All of these QoL features are present in the i3 firmware.

JohnnyDeer, 2020-06-24:
Hi @keksa , filament loading function is improved in FW 4.1.0-RC1. Please try it out. I think we can consider this issue as solved, do you agree?

keksa, 2020-07-06:
Hey @JohnnyDeer, it's way better, thank you, closing.

closed 2020-07-06 by keksa