prusa3d/Prusa-Firmware-Buddy #261: pause not immediate + resume print after filament sensor interrupt

keksa, 2020-02-29:
I had an issue, where the sample filament snapped during print. I have the filament sensor, so this was detected and the print paused.

I reloaded the filament and resumed the print. Here I've encountered 2 issues:
1) The printer made a small blob on the spot on the layer where the interrupt / resume happened. I'm not sure if this was before the interrupt or after resuming the print, as it's very much possible I just did not notice it. But I think it was after the resume. It might be nice if the printer did a line on the side (exactly like when starting a print) as it might eliminate this kind of issue and/or give me time to notice and resolve it myself.

2) Once I noticed the blob, I immediatelly pressed the pause button. Sadly the pause is not immediate, so the printer wanted to finish the layer. This caused the nozzle to hit the small blob and pop the print off the bed, resulting in a failed print. IMO the pause should be immediate, since I think the only time I would want to pause the print is to handle a print issue and finishing the layer might only make the issue worse (as it did in my case).

loglow, 2020-03-02:
I can confirm that I've also encountered very similar issues regarding the extruder oozing and blobs forming during and after pausing a print.

Is this an inherent limitation with the technology, or could the issue be addressed via automatic nozzle temp adjustment, filament retraction, and/or nozzle wiping before resuming?

Regarding the pause not being immediate, I'm not sure it's possible. The g-code commands are buffered before being executed, and the commands already in the buffer all get executed before the pause actually takes effect.

I'd be curious to hear from any developers if interrupting these already-buffered commands would be conceivable or not (via software).

JohnnyDeer, 2020-06-09:
Hi, as @loglow wrote, commands are in buffer and is not possible to stop immediately. Retraction may help, but that is about create feature request and then internal discussion.

JohnnyDeer, 2020-10-21:
Hi, we decided to not make changes in this case. Thank you for understanding.

closed 2020-10-21 by JohnnyDeer