prusa3d/Prusa-Firmware-Buddy #263: Firmware crashed when started heating w/o heatbed

tmeinlschmidt, 2020-03-02:
firmware 4.0.2

Steps to reproduce
- take off the steel plate
- start heating (PLA is enough)
- wait until temperature reaches 60oC for the bed
- place steel plate

After a while screen goes red with "Bed" text and nothing else. Rotary control works, icons will appear, but for example Settings/Temperature will lead to complete freeze.

[image]

julianrupp, 2020-03-20:
Slightly different outcome for me. Printer will spin up all fans to 100% and freeze with a safety shutdown warning message that forces you to reboot.

I understand this "feature" is in place to detect broken temperature sensors before that could lead to severe issues like heater cartridge overheating, but it obviously does not have enough leeway for such a common and completely normal usage scenario.

The i3 has this feature as well, but it has never triggered yet in such a scenario for me - better sensitivity calibration there!

JohnnyDeer, 2020-05-28:
Hi, @julianrupp you are right, this behavior is on purpose. Mini have different hardware than i3 and can't be tuned as sensitive as i3. Safety first, so i will close this issue with conclusion, that this behavior is feature.

closed 2020-05-28 by JohnnyDeer