prusa3d/Prusa-Firmware-Buddy #264: Resuming print not cleaning noozle

tmeinlschmidt, 2020-03-02:
firmware 4.0.2

steps to reproduce:
- start print
- pause
- wait until some filament starts exiting the noozle
- resume print

expected behaviour:
- noozle is cleaned up outside printing area with simple short line

Bertware, 2020-05-14:
This currently causes paused prints to fail, as the nozzle will ooze onto the print, causing a blob and failed prints as a result.

isopterix, 2020-06-11:
Yes, please add this!!! I just ran into the same issue... luckily I was quick enough to take out the blob with some sizzlers... Otherwise it would have been a complete fail.

Bertware, 2020-06-11:
A special case would be a print that is paused multiple times. In this case, the printer should either stack the "wipe lines" or find another unused area to wipe the nozzle. In the meanwhile, a fix where this only happens for the first pause of a print would be nice already!

Eldenroot, 2022-12-08:
If not #2760 then we need this one. This could save a lot of time and avoid issues because of pausing

closed 2024-07-04 by github-actions[bot]