prusa3d/Prusa-Firmware-Buddy #265: USB still not 100 % ok

Pitel, 2020-03-02:
I still have to go to some menu section and back to the main screen when reinserting the USB drive. Otherwise, the _Print_ menu item is disabled.

Lino77, 2020-03-06:
I still have print on the display, even though the stick was pulled out. not always, but now and then

runemart, 2020-03-24:
Confirmed here, still an issue (using the USB key I got with the Mini). 
USB icon in top right corner changes state (bright vs dim) correctly, but Print menu icon does not. 
Video of the behaviour: https://www.youtube.com/watch?v=6rMDdcDB7Tc

x10102, 2020-03-25:
Confirmed here also.

kdpainter, 2020-03-26:
I have seen this as well.

0xygen8, 2020-04-05:
The Original USB stick works 99% of the time other Intenso USB 3 sticks only work when the Mini is booted with the stick inserted. Inserting Intenso (and others) when Mini is booted fail 95%  of the time

Pitel, 2020-04-07:
We can close this in favor of #21.

closed 2020-04-07 by Pitel