prusa3d/Prusa-Firmware-Buddy #269: email or call http resource after finishing print

MichaelReitsma, 2020-03-03:
It should be able to email when finished.
Use some additional information such as filename, filament type, duration etc. to add informative text in the email.
This could also be used to report problems or errors.

An alternative way is to be able to use a service like pushbullet. That would show like a popup on your phone or anywhere you have a pushbullet client installed.

See also https://docs.pushbullet.com/#create-push .

Isaac12x, 2020-04-12:
You can do that with Octoprint :)

danopernis, 2024-05-17:
While this issue is certainly older than https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3826 I am still closing this one in favour of the newer one simply based on number of comments. Eventual discussion can continue there.

Just to note, email support is almost definitely not happening. It might be worth to consider [Zawinski's Law](https://en.wikipedia.org/wiki/Jamie_Zawinski#Zawinski's_Law) :smile:

closed 2024-05-17 by danopernis