prusa3d/Prusa-Firmware-Buddy #280: first layer calibration - wrong temperature

keksa, 2020-03-07:
FW 4.0.4

The in-built first layer calibration tool is doing mesh bed leveling with nozzle temperature of 215 °C. Since regular prints do mesh bed leveling with nozzle temperature of 170 °C and then wait up for higher temp, I think this might be a bug. 

This also results in random blobs of filament on the bed before the calibration actually starts.

Isaac12x, 2020-04-12:
I think this one is solved with the 4.0.5 update. Can you check?

keksa, 2020-04-26:
Hello @Isaac12x, just updated to 4.0.5 and checked. Nothing changed, still the same behaviour.

JohnnyDeer, 2020-05-05:
Hi @keksa , thank you for reporting, but this issue is already reported. I'm closing this one as duplicate of #157

closed 2020-05-05 by JohnnyDeer