prusa3d/Prusa-Firmware-Buddy #282: Printing “completes” early, leaving hotend stopped at printing temperature in the middle of the print

laposheureux, 2020-03-08:
I’m using gcode generated with PrusaSlicer that has successfully printed multiple times while running on 4.0.0 and 4.0.1, but it seems to fail on 4.0.3 (my first test today) and 4.0.4 (updated to this after the first failure) in a dangerous way.

The initial print time shows an expected 5-and-change hour print (both expected and correct from past prints), and two attempts today have “finished” after around 4 hours and 3.5 hours as if complete, but with the bed and hotend still hot. The screen shows a full progress bar, the reprint button available, and an elapsed time that doesn’t match how long the print should have taken.

It appears that without intervention the printer will stay at full printing temperature indefinitely. I will attach the gcode soon.

laposheureux, 2020-03-08:
[image]

Here is a photo showing the “end state” of the first print, completed before it should have with the hot end sitting at 210.

laposheureux, 2020-03-08:
Other observation: even after I told it to go to cooldown and went back to home, after it finished cooling down it left the hotend fan on. I just came back to it hours later and noticed it was still blowing full speed. Here is the gcode that has had these failures: https://www.dropbox.com/s/5tsq0kj0qvuunxo/Edge%20Main%20Pose.gcode?dl=0

frankincell, 2020-03-23:
I had this occur with a corrupt gcode file.  I had printed the same item a few times from that gcode file and then suddenly it would stop at 75% done and just sit there cooking.   I tried copying the gcode file onto the USB drive again but it did the same thing.  Then I deleted the gcode file, copied a picture to the USB drive, and copied the gcode to the USB again.  From then on it would print perfectly.  Either there's a bad memory spot on the USB drive, or the gcode file became corrupt and copying it over again in Windows didn't fix the corruption.   Give it a try and let us know if it fixed your problem as well.

JohnnyDeer, 2020-05-25:
Hi @laposheureux , can you please upload Gcode you mentioned? Did this happen with more gcodes, or just with this one? Did you tried it out on 4.0.5?

laposheureux, 2020-06-29:
It was just the one and the gcode is linked above in a Dropbox shared link. I haven't retried it with the latest firmware yet.

JohnnyDeer, 2020-06-30:
@laposheureux thank you for info. Unfortunately i can't reproduce it. Please try it with 4.1.0-RC1.

JohnnyDeer, 2020-07-28:
@laposheureux do you still have this problem with FW 4.1.0?

precision, 2020-09-10:
I'm not 100% sure, but I think I am seeing the same thing via USB printing on 4.2.1.

Twice since upgrading to 4.2.1 I've printed a GCODE file it stopped mid-print with motors still active and heaters heating.  Cancel the print and send it again and it works fine.

I saved a crash dump the last time it happened via the LCD if that would be of any help.

JohnnyDeer, 2021-02-16:
@laposheureux is this issue still actual?

JohnnyDeer, 2021-08-02:
This issue will be closed soon for no responses.

closed 2021-08-23 by JohnnyDeer