prusa3d/Prusa-Firmware-Buddy #283: Pause print and wait for user input to continue

RobeeeJay, 2020-03-08:
Hi lovely people! I've been trying to pause a print and wait for human input to trigger a continue, but there doesn't seem to be a way to do this.

M601 doesn't seem the right code to use as it doesn't ask for user input from what I can tell, and isn't in the firmware anyway from what I can see.

M1 seems to do nothing but make it stuck in the printing UI, unable to pause or stop, or exit, without hitting reset.

M226 seems like it could wait for the encoder button by waiting for it's pin state, but I can't work out what pin that would be and what state it should be in when not pressed.

M125 looks like it does what I need but only when `SDSUPPORT` is enabled, which I don't think it is on the mini so `wait_for_confirmation` is never called because I guess `sd_printing` is always false. I really don't want to invalidate my warranty to install a custom build to try this. :(

Am I missing something obvious?

CyberHugger, 2020-03-26:
The M0 command does allow you to specify a wait time in milliseconds or seconds but you should also be able to cancel the pause by user action (i.e. click the encoder button). The pause is working on the MINI but there is no apparent way to cancel the pause using the control. This does work on the MK3S.

RobeeeJay, 2020-03-26:
I might be closing this soon, because the new RC firmware looks like it has a proper pause in there now!

OndraCerv, 2020-03-27:
Hi, 

I have firmware 4.0.5 RC1 and PrusaSlicer 2.2.0. 
M601 code not stopped and printer continue in printing.

RobeeeJay, 2020-03-27:
Looking at the code in the 4.0.5 branch, M601 isn't in there, and the code for M125 is still unchanged, so this looks to still be broken. 😥

Eisfoehniks, 2020-04-11:
Same, have prusaslicer 2.2.0 and firmware 4.0.5. Adding a "Pause" in prusaslicer doesn't work on the MINI. It will just continue printing. As a workaround I use the "Color change" and insert the same filament again.

JohnnyDeer, 2020-05-14:
Hi, at this moment M601 is not supported by Mini. We are working on implementation, M601 will be working in next FW version. Thank you all for reporting and patience.

martin8425, 2020-06-21:
Can anybody confirm this is working in 4.1.0-RC1? It's not specifically mentioned in the release notes... ;)

RobeeeJay, 2020-06-21:
I have bad news, it's not working in RC1. 😞 I tried a test print both via Octoprint and USB stick. M601 is in the Gcode (I checked) but the printer ignores it.

liyanage, 2020-07-01:
Looking forward to this getting implemented, I just asked about this on the Prusa forums and was directed here: https://forum.prusaprinters.org/forum/general-discussion-announcements-and-releases/pause-print-g-code-ignored-during-print-on-prusa-mini/#post-226090

JohnnyDeer, 2020-07-30:
Finally I can confirm that M601 is implemented in latest FW 4.2.0-RC1. Please try it out and confirm it too :)

RobeeeJay, 2020-07-31:
Does this not work from Octoprint? I just created a simple cylinder in PrusaSlicer, added a layer change, edited the gcode and replaced M600 with M601, uploaded it to Octoprint, and it didn't pause at all. 😞

JohnnyDeer, 2020-08-05:
Most important for us are bugs and features on normal printing. Printing via OctoPrint have kinda low priority at this moment so only important bugs will be repaired (filament sensor function, Z moving while live adjust, etc.). Thank you for understanding.

gabi8es, 2020-11-18:
I have just discover this problem with octoprint, the M601 is not working from octoprint, and the support from USB is limitated because the message introduced in the prusa slicer doesn't appear in the screen.
In the same way, if you use the M600 to a filament change the is it no possible to introduce any message to inform the user about the new color needed. could this will take in consideration in order to include in a new firmware version?
Thanks in advance

foorschtbar, 2021-01-15:
It still (Jan 2021) not working with current FW and OctoPrint. Any china printer under 200 bucks could stop printing with a G-Code command and OctoPrint, but a Original Prusa not? You can do better than that guys!

foorschtbar, 2021-01-15:
> Most important for us are bugs and features on normal printing. Printing via OctoPrint have kinda low priority at this moment so only important bugs will be repaired (filament sensor function, Z moving while live adjust, etc.). Thank you for understanding.

IHMO thats not a feature, its a basic function thats missing. Any china printer under 200 bucks could stop printing with a G-Code command and OctoPrint, but a Original Prusa not? You can do better than that guys! I need no UI for that at the printer, continue with the WebUI is okay for me, but simple ignore the commands is very bad 👎

RobeeeJay, 2021-01-15:
I think you can work around this on Octoprint, by adding M601 to the list of `Pausing commands` under the `Firmware and protocol` tab.

Also, make sure you add this code to `Before print job is resumed` in the GCODE section otherwise you will be printing thin air. 😞

Duckle29, 2021-01-26:
Just wasted a couple of hours and some plastic because of this. At the very least put a warning in PrusaSlicer if you have a MINI configured for octoprint selected :(

skarfie123, 2021-08-15:
tried this GCode from USB and it didn't work:
```
G28 ; home all without mesh bed level

; move to centre
G1 Z5 F300
G1 X90.000 Y90.000 F900

G1 Z10 F300

M117 Pause
M601

G1 Z20 F300
```

coaxial, 2021-09-28:
This still didn't work a year later in the latest firmware version. 

At least put a warning in the slicer because I too wasted filament for a print that didn't pause.

your-friend-alice, 2021-11-03:
Honestly, this is wild. This product should not have shipped without this. The mini would've been OK if network printing was supported (as implied by the presence of an ethernet port), but of course that hasn't happened either. Really disappointed, because my Prusa Mini has been so solid, but without this basic functionality I really can't recommend this printer to anyone.

Especially weird since color change works, suggesting that this would be a trivial fix that just hasn't been prioritized. Is the prusa team at least receptive to PRs from new people? If this takes much longer I may have to void my warranty and try my hand at fixing this myself

RobeeeJay, 2021-11-03:
Have you tried this Alice? https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/283#issuecomment-761168887

coaxial, 2021-11-03:
I have, and it didn't do anything. 

The bottom line is that this is a (2 year old!!) bug that Prusa refuses to fix because anything Octoprint is a "nice to have" rather than a core feature... for a printer that is being marketed as a printing farm printer.

your-friend-alice, 2021-11-05:
@RobeeeJay Just tried as well, it partially worked. I added `M601` to the list, and restarted my serial connection. The pause worked, however the head just stopped in place and sat there at full temperature, which is not ideal since it'll be slowly cooking the filament sitting in the head as well as anything near the head, and blocking access to the print. Note: this partially-working pause can also be reached by adding Octoprint's `@pause` code to a layer through the "custom g-code" prompt in PrusaSlicer. The basic problem remains though; we can get Octoprint to pause sending commands to the printer, but the printer is still ignoring M601s.

I guess the next step to a full workaround is adding custom G-Code to the "After print job is paused" and "Before print job is resumed" fields in Octoprint to replicate the printer's built-in M601 behavior. That's gonna be a bit tedious but doable.

Note to Prusa devs, I'm willing to hack together a workaround for myself, but this is still a deal-breaking bug and I still will consider this product unshippable until its basic functionality works out of the box.

Firmware version: 4.3.3, Octoprint version: 1.7.2.

RobeeeJay, 2021-11-05:
@your-friend-alice Here is my pause code if it helps:

M117 Print Paused  ; Comment
G91 Z              ; Set to relative positioning mode
G1 Z20             ; Move hotend away from part by 15mm
G90 Z              ; Set to absolute positioning mode

M117 Print Resumed ; Comment
G91 Z              ; Set to relative positioning mode
G1 Z-20            ; Return to original position
G90 Z              ; Set to absolute positioning mode
M83                ; Set extruder to relative

your-friend-alice, 2021-11-06:
@RobeeeJay thanks! this is also really helpful. The one thing I'll want to improve is move Y to 180, which will be tricky, since I can't cheat it with relative positioning then. I remember seeing commands for storing and recalling positions, I'll dig it up and try it

your-friend-alice, 2021-11-06:
Alright, here's what I've got:

## After print job is paused:
```
G60 ; Save current position
G91 Z ; Set to relative positioning mode
G1 Z20 E-3; Move hotend 20mm up from part
G90 Z ; Set to absolute positioning mode
M83 ; Set extruder to relative
G1 X180 Y180 ; Move the bed to the front
```

## Before print job is resumed:
```
G61; Return to original position. I cannot find any way to have it first move in XY, and then Z and E, despite this command having flags documented to support this.
```

Other notes: `M125` looks like a better option than all the code in "after print job is paused", but it doesn't appear to work.

coaxial, 2021-11-06:
If this helps anyone, this is my "after print job is canceled" script: 

```

G91; Set to Relative position
G1 E-1 F300; retract the filament a bit before lifting the nozzle
G0 Z80; move z axis up 15
G90; Set to Absolute position
G28 X; home X axis
G28 Y; home Y axis
G1 Y180 F5000; move part out for inspection
G91; set back to relative position
M104 S0 ; turn off extruder heat
M140 S0 ; turn off heated bed
M106 S0 ; Turn off fan
M18; Move Freely
```

After print job is paused:

```

G91; Set to Relative position
G1 E-1 F300; retract the filament a bit before lifting the nozzle
G0 Z80; move z axis up 15
G90; Set to Absolute position
G28 X; home X axis
G28 Y; home Y axis
G1 Y180 F5000; move part out for inspection
G91; set back to relative position
G4 ; wait
```

foorschtbar, 2021-11-08:
> Especially weird since color change works [...]

This.

jjxxs, 2021-12-30:
I just experienced this on a mk3s+ running over serial with octoprint. I have used PrusaSlicer to insert two pauses in the print, one roughly after layer 50, the other around layer 650. The first pause worked fine. The second one was ignored. I verified that the commands are in the gcode:

```
G1 E-0.04000 F2100.000
G1 Z5.100 F10800.000
;AFTER_LAYER_CHANGE
;5
;PAUSE_PRINT
M117 Insert
M601

; ...

G1 E-0.04000 F2100.000
G1 Z204.900 F10800.000
;AFTER_LAYER_CHANGE
;204.8
;PAUSE_PRINT
M117 Insert
M601
```

The print took half a day and a whole lot of plastic went to the bin... Prusa, please fix this.

Gordontgopher, 2021-12-31:
> I just experienced this on a mk3s+ running over serial with octoprint. 
This is an issue report for the Prusa Mini, which runs different firmware to your Mk3s+. Suggest you check the Mk3 fw repository to see if there is an open issue there.

coaxial, 2022-02-05:
Here is to hoping that in 2022 (2 years later), that bug will finally be fixed.

moggieuk, 2022-02-15:
Add my name to the list... Prusa, your slicer allows you to define pause points and inserts M601 into the gcode.  Even has cool functionality to allow you to customize the M117 message but your printers have to support it....   It's disappointing how long this is taking to support on the Mini.  I just built a Voron 0.1 and added pause/resume support in a couple of hours.

meermanr, 2022-09-20:
Quoting https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/283#issuecomment-899026594

> tried this GCode from USB and it didn't work:
>
> ```gcode
> G28 ; home all without mesh bed level
>
> ; move to centre
> G1 Z5 F300
> G1 X90.000 Y90.000 F900
>
> G1 Z10 F300
> 
> M117 Pause
> M601
>
> G1 Z20 F300
> ```
>
> This is supposed to be a very simple gcode to test the M601 works. Moving to the centre and pausing while moving upwards.

Just to confirm this is still the case in Firmware 4.3.4 when printing **from USB**. So this issue is not limited to serial printing.

----

In case this is useful to others, I have worked around this using `M109` to set _and wait_ for a target temperature to be reached, and setting that temperature lower than room temperature:

```gcode
; Cool nozzle
M117 Paused. Change nozzle and set temp to resume.
M109 R1 ; Set **and wait** for target nozzle temperature
```

This prints a message (via `M117`) and then set-and-wait nozzle temperature (`M109`) to reduced temperature of 1ºC (`R1`). Note that the more common `S` for setting temperature will only wait for heating, not cooling. Cf. [Marlin firmware: M109 - Wait for Hotend Temperature](https://marlinfw.org/docs/gcode/M109.html).

When I want the print to resume I use the "Tune" menu on the Prusa MINI (which is available even during serial printing) to set the nozzle temperature at or above the current ambient temperature. (My hand-crafted GCode will restore the proper nozzle temperature itself, so I don't need to remember it: I need only get the print to resume)

In my case I want to change the nozzle during the print, using E3D Revo quick-change nozzles which can be changed when cold using just your fingers. Therefore I can't use filament change (`M600`) as that would keep the temperatures up, and set off the thermal runaway alarm when I remove the nozzle.

napter, 2022-09-21:
I was going to open an issue because M0 \ M1 doesn't work but it seems it is mentioned here. Why doesn't M0 work? It is supposed to pause and wait for the user to press a button. It works on the MK3S.

Prusa-Support, 2022-09-24:
Thank you all for your input and for your participation.

I'm sorry the MINI firmware at the moment doesn't support such codes yet.

Our developers are informed about this situation but, as some of you may know, they are truly busy with the optimization of Prusa Link and other priority tasks, and please notice Octoprint is a 3rd party software.
All of these issues are noted down and our developers will be more than happy to look into these at their earliest convenience, possibly with as little impact as possible on the current super-tight development plans.
Indeed, I don't want that some firmware polishing for a better serial/Ocorpint compatibility would pose a conflict to the short-term development plans (I can't really tell, just "wild guessing").

Please freely keep on reporting updates and suggestions. The issue won't be ignored.
Our developers are working for you. Thank you for your support, patience and understanding. 

Michele Moramarco
Prusa Research

napter, 2022-09-24:
> Thank you all for your input and for your participation.
>
> I'm sorry the MINI firmware at the moment doesn't support such codes yet.
>
> Our developers are informed about this situation but, as some of [...]

I understand. But to be clear this is not related to octoprint specifically. This is an issue with USB printing. The missing feature is part of Marlin and it works in the MK3S (it also works on every other Marlin based printer that I have). 

I suspect a large number of your users print that way. Personally, I don't think fixing USB printing is equivalent to 'resolving an issue with third party software'. This is an issue with the printer firmware when USB printing, not an issue with third party software. Another example is how the runout sensor doesn't work when USB printing if the lcd isn't on the front screen. Another is that the change filament menu option is disabled when connected via USB. I feel like printing by a physical USB drive should not be the only supported method. 

Thank you for the response 🙏. I truly appreciate it. I hope they'll put some focus on usb (cable) printing at some point as it is the only reasonable way to manage a print farm of prusa minis.

coaxial, 2022-09-24:
> put some focus on usb (cable) printing at some point as it is the only reasonable way to manage a print farm of prusa minis.

Even more so when the Mini was (and still is) sold as a "print farm workhorse". And yet, three years later, still has so many issues related to features that are table stake for even the entry level Chinese printers. Ditto the filament sensor that is made by Prusa and sold as an option on the Mini yet still not quite fully functional. 

Saying that serial printing is third party software and so not really Prusa's concern or priority is disingenuous at best. Octoprint is at the heart of many (most?) print farms. 

I'm sympathetic to the challenges of a deep backlog and finite engineering resources, but I paid good money for the printer and feel quite let down in many respects about it. It's one thing to take a few months to fix a bug or implement a feature, but three years and counting to make good on your marketing material? I can't help but feel taken for a ride.

meermanr, 2022-09-26:
The pause gcode doesn’t work when USB Printing, so it doesn’t appear to be implemented _at all_. 

Serial printing is secondary concern for me (though can pause whatever is driving the serial port of I must)

Sandiegoo, 2022-10-07:
Wasn´t this fixed years ago? And also was published on twitter, that the M601 Code works now on Prusa Mini.
https://twitter.com/prusa3d/status/1291042582366167043?lang=de

I just found out it still doesn´t work with the latest firmware (4.3.4) :(

It would be much better then the color change G Code.

I hope this will be implemented in the next version of prusa mini firmware.

coaxial, 2022-10-07:
@Sandiegoo it says from *USB* which is all that matters apparently because anything serial is a nice to have and not a core feature according to previous Prusa comments.

meermanr, 2022-10-07:
But it **doesn’t** work from USB either!

Prusa-Support, 2022-10-12:
As per the firmware release [Firmware-Buddy v4.2.0](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.2.0), the code M601 was fixed for use via USB flash drive (reading from USB drive is a basic function of the standalone MINI printer) but it is not fixed (yet) for use over serial communication (OctoPrint, Pronterface...) - which is not the designed way to use the printer but useful to unlock other cases of use.
As far as I can tell, after that release, the printer will no longer ignore an M601 code mid-print, if such code is included in a g-code file, being run locally via USB stick.

As a company, I find it legit for us to follow our main development plans but that doesn't mean the requests from the community will be ignored in any way.

In other words, quoting @coaxial, it is safe to say this is not a core feature, but we will work on it.
I'm sorry about the persisting problem. This issue will stay open.

Michele Moramarco
Prusa Research

coaxial, 2022-10-12:
I was quoting prusa's statement that anything serial isn't part of the core set of functionality and just a "nice to have".

I and several others here strongly disagree that serial support for a printer marketed as a "print farm workhorse" is a nice to have and optional. As said before, full gcode support over serial and usb are table stakes for most other printers including the cheaper Chinese ones.

It's hard to understand how the Mini is different and doesn't fully implement serial printing.

Prusa-Support, 2022-10-19:
Of course, the printer hardware supports the serial connection and the firmware is open source. Users and community can make use of these characteristics to their advantage, according to their specific needs.
The community's contribution and suggestions are always very appreciated.

Please notice the issue will stay open for our developers to look into this at their earliest convenience, consistently with the company development plans. I'm afraid I can't further argue without sounding redundant at this point and I hope you can understand.
I'll keep monitoring and taking relevant notes for our developers. Thanks.

Michele Moramarco
Prusa Research

coaxial, 2023-03-03:
It's been three years (in 5 days) that the issue has been open, looking forward to a fix to this bug.

napter, 2023-03-03:
> It's been three years (in 5 days) that the issue has been open, looking forward to a fix to this bug.

Apparently "serial communication is not the designed way to use the printer" and so all the bugs and quirks that don't work right when printing over serial are not a priority. The Prusa Product Manager expects everyone to use USB drives since printing over a usb cable is "not a core feature".

Prusa-Support, 2023-07-05:
I understand. Thank you again for your input.

On the bright side, Prusa MINI will be available for Prusa Connect in the near future, enabling advanced print farm control and monitoring features.

Michele Moramarco
Prusa Research

napter, 2023-07-05:
> I understand. Thank you again for your input.
> 
> On the bright side, Prusa MINI will be available for Prusa Connect in the near future, enabling advanced print farm control and monitoring features.
> 
> Michele Moramarco Prusa Research

I don't mean to offend but I'm not sure you do understand. The response has been consistently that this, and all the other USB printing issues aren't going to be addressed as they are an octoprint issue. This is a miscategorization as the issues are USB issues. For some reason, which has not really been addressed, USB printing has severely reduced functionally from the display. On the other hand this functionally does exist and works very well on the mk3s when usb printing. If there was a reason USB printing is not going to get any fixes, I would understand but saying it's an octoprint issue isn't right. 

I hope you take this constructively, I love you guys and the work you do

coaxial, 2023-07-06:
This is tone deaf. People are saying over and over that full serial functionality is not optional nor nice to have, you keep telling us how fabulous Prusa connect will be and that implementing all features related to serial printing is low priority. 

We. Don't. Want. Prusa. Connect. 

Some of us prefer to use the printer with the battle tested, stable, very popular Octoprint. Why does it take years for nothing to happen on this and other serial-related tickets??

tetsudou, 2023-07-06:
The problem with Prusa Connect is that it's cloud-based. For our company
that is not acceptable due to intellectual property reasons, which is why
it is not really an option.

Prusa-Support, 2023-09-27:
Octoprint is not a product of ours but we acknowledge that many members of our community make use of it.
We never meant to suppress it - nor any other serial-line-based application - and we never denied fixing related issues.

The internal development plans have been quite packed so far but we can gladly affirm that all of the effort is finally paying back.

Now our developers plan to run an overhaul of Octoprint and serial connection issues starting with the upcoming firmware version - https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/189#issuecomment-1731224779.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-12-27:
Since [FW 5.1.0](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0) it should be possible for Prusa Firmware and Octoprint to seamlessly work together.

>To make sure everything works correctly, you need to update the Octoprint profile with the following G-codes (in the GCODE Scripts section):
>- After print job is cancelled: M604
>- After print job is paused: M601
>- Before print job is resumed: M602

https://blog.prusa3d.com/prusa-autumn-news-summary-2023-new-firmware-prusaslicer-mk3-5-info-and-more_86935/

Please test the most recent available firmware version and feel free to share feedback.
Can this issue be closed?

Michele Moramarco
Prusa Research

coaxial, 2023-12-28:
I have tried pausing, resuming, and canceling in Octoprint with firmware 5.1.2 and the above settings on the Mini and it does seem to finally do the right thing. Thank you for the update, it is a welcome addition even though it took way too long to be acknowledged and finally be implemented.

Prusa-Support, 2024-01-26:
It was not our intention to discard Octoprint, but the support for it was not natively in our development plans. We recognize though that Octoprint is very popular and appreciated by our community and took action accordingly.

I'm sorry that it took so long and I thank you for your feedback.
This issue will be closed as completed.

Michele Moramarco
Prusa Research

closed 2024-01-26 by Prusa-Support