prusa3d/Prusa-Firmware-Buddy #303: Hotend Fan

Lino77, 2020-03-19:
can someone please explain to me where you limited the hot end fan. the fan only runs at approx. 75%, which is probably sufficient for your mini with your delta fan, but I have rebuilt an MK3 with your buddy board, and I can't find the setting for it anywhere. can anyone help me

Lino77, 2020-03-19:
gefunden

closed 2020-03-19 by Lino77

PeterDHabermehl, 2020-10-17:
> can someone please explain to me where you limited the hot end fan. the fan only runs at approx. 75%, which is probably sufficient for your mini with your delta fan, but I have rebuilt an MK3 with your buddy board, and I can't find the setting for it anywhere. can anyone help me

I've got the same issue, could you share your wisdom, please?