prusa3d/Prusa-Firmware-Buddy #304: [BUG] hwio_ai3des_2209_02.c line 677 Extruder Fan

Lino77, 2020-03-19:
line 677 _hwio_pwm_analogWrite_set_val(HWIO_PWM_FAN1, ulVal ? 100 : 0);
overwrites the Configuration_A3ides_2209_MINI_adv.h  value
EXTRUDER_AUTO_FAN_SPEED
Change line to 
_hwio_pwm_analogWrite_set_val(HWIO_PWM_FAN1, ulVal ? _pwm_analogWrite_max[HWIO_PWM_FAN1] : 0); 
then the values from Configuration_A3ides_2209_MINI_adv.h are accepted again

JohnnyDeer, 2020-06-02:
Hi @Lino77 , Marlin does use software PWM. This is workaround to use hardware PWM without modification of Marlin.

closed 2020-11-07 by Lino77

PeterDHabermehl, 2020-11-15:
> Hi @Lino77 , Marlin does use software PWM. This is workaround to use hardware PWM without modification of Marlin.

So, @JohnnyDeer , how can I change the PWM output for the fan in order to get 100% PWM (5V) to the Noctua I would like to use? Thanks for your help!

MinnesotaMean, 2022-03-29:
> > Hi @Lino77 , Marlin does use software PWM. This is workaround to use hardware PWM without modification of Marlin.
> 
> So, @JohnnyDeer , how can I change the PWM output for the fan in order to get 100% PWM (5V) to the Noctua I would like to use? Thanks for your help!

Yes please how do we do this! Thank you.