prusa3d/Prusa-Firmware-Buddy #309: Bed leveling - first layer thickness not constant

ProdYJ, 2020-03-20:
At almost every new print, I have to tune the first layer with the live z offset. Sometimes the printer prints a bit too low, sometimes a bit too high.

0xygen8, 2020-03-20:
I have seen that if you slice it yourself it is more or less constant printing from a 3mf file does not honor your live z value (I have very deep scratches in my pei sheet)

ProdYJ, 2020-03-21:
I really have no idea how the 3mf file can have any influence on the z offset of the printer, I checked and there is nothing in the gcode to do that.

x10102, 2020-03-25:
Have you checked your M.I.N.D.A height?

ProdYJ, 2020-03-26:
My M.I.N.D.A. is well in place and doesn't move. May its height, even if it does not change, cause this problem?

dahuafschmied, 2020-03-26:
i have the same Problem. Mostly with a freshly switched on and cold printer. (21°C room Temp)

madelponte, 2020-03-26:
I have pretty much solved this on my printer by adding a custom wait gcode (G4 S300) for 5 min right after the wait for bed temp. The printer should wait longer after the temperature is set before doing the mesh bed leveling

ProdYJ, 2020-03-26:
I'm really not sure of what I'm saying here, but I think I have red that the M.I.N.D.A. has a thermistor builtin to compensate the influence of its own temperature on its measurements. 
The code to compensate this maybe has an error? I'm not skilled enough to verify this myself.

dinands, 2020-03-29:
I have the same problem here I have to lower the Z a little bit started at 0.365 is now at 0.430 this does not seem right to me. went back to 4.0.4 and no problem back to 0.365

joeynuggetz, 2020-04-04:
> 4.0.5 I have the same problem here I have to lower the Z a little bit started at 0.365 is now at 0.430 this does not seem right to me. went back to 4.0.4 and no problem back to 0.365
> 
> back at 4.0.4 and yet again the problem that the Z expires is now at -575.
> 
> gone back to 4.0.3 factory reset now Z to -460 and after div testing it now remains constant

I've seen this problem and simply resetting the printer or restarting the print fixes the issue. Not sure if you going back to a diff firmware is really a fix. Would be curious if you went back to the firmware with the problem to see if it gets screwy again. I've been dealing with this first layer crap since day one.

dinands, 2020-04-04:
> I've seen this problem and simply resetting the printer or restarting the print fixes the issue. Not sure if you going back to a diff firmware is really a fix. Would be curious if you went back to [...]

the problem is the winda. when it is warm the problem is gone but with a cold start the life Z is too high when I stop the print and immediately print again the problem is not there.
it really has to do with the temperature of the winda. I have modified the start G.code now no more problems

dennisscholing, 2020-04-06:
Same issue here. First layer is only consistent when editing te start G Code to warm up the MINDA. 
Tried several firmware versions, other MINDA and factory resets. Problem suddenly occurred after update 4.0.5. Never had this issue before.

OndraCerv, 2020-04-07:
Hi,

i have firmware update 4.0.5 (yesterday updated). i have same issue with new firmware. Before print i have first layer calibration. In print first layer i need manual offset z axis + - 250um.

dahuafschmied, 2020-04-07:
Does anyone know if any of the developers are reading this bug tracker or the forum?

dinands, 2020-04-07:
with support they always say that the winda is not good, but that is not the problem.
I don't know if the developers know about the problem

dahuafschmied, 2020-04-07:
> with support they always say that the winda is not good, but that is not the problem.
> I don't know if the developers know about the problem

i know. got a new MINDA from support. ;-) did not fix the problem....

OndraCerv, 2020-04-07:
to dinands:

you write: ( I have modified the start G.code now no more problems
Now running on 4.0.5
(the Gcode puts the head in the middle of the bed during warming up and then waits 80sec, before leveling the bed)
if you have already loaded filament you have to wait longer 120sec + / _)). 

Can you me write this the G-code?  I will tested you solution. 

Thanks

dinands, 2020-04-07:
> to dinands:
>
> you write: ( I have modified the start G.code now no more problems
> Now running on 4.0.5
> (the Gcode puts the head in the middle of the bed during warming up and then waits 80sec, before [...]

this is my start gcode
G90 ; use absolute coordinates
M83 ; extruder relative mode
G28 ; home all
G1 Z50 F720 ; park position
M104 S170 ; set extruder temp for bed leveling
M140 S[first_layer_bed_temperature] ; set bed temp
M109 R170 ; wait for bed leveling temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
G4 S120; wait 2 mins
G28 ; home all without mesh bed level
G29 ; mesh bed leveling 
M104 S[first_layer_temperature] ; set extruder temp
G92 E0.0
G1 Y-2.0 X179 F2400
G1 Z3 F720
M109 S[first_layer_temperature] ; wait for extruder temp

; intro line
G1 X170 F1000
G1 Z0.2 F720
G1 X110.0 E8.0 F900
G1 X40.0 E10.0 F700
G92 E0.0

M221 S95 ; set flow

dinands, 2020-04-07:
I'm not home now send it tonight

OndraCerv, 2020-04-07:
dinands:

thanks you very much

dennisscholing, 2020-04-07:
> to dinands:
>
> you write: ( I have modified the start G.code now no more problems
> Now running on 4.0.5
> (the Gcode puts the head in the middle of the bed during warming up and then waits 80sec, before [...]

`
G90 ; use absolute coordinates
M83 ; extruder relative mode
G28 ; home all
G1 X100 Y100 F4000
G1 Z1.5 F50 ; park position
M104 S170 ; set extruder temp for bed leveling
M140 S[first_layer_bed_temperature] ; set bed temp
M109 R170 ; wait for bed leveling temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
G4 S80; wait 2 mins
G28 ; home all without mesh bed level
G29 ; mesh bed leveling 
M104 S[first_layer_temperature] ; set extruder temp
G92 E0.0
G1 Y-2.0 X179 F2400
G1 Z3 F720
M109 S[first_layer_temperature] ; wait for extruder temp

; intro line
G1 X170 F1000
G1 Z0.2 F720
G1 X110.0 E8.0 F900
G1 X40.0 E10.0 F700
G92 E0.0

M221 S95 ; set flow
`

OndraCerv, 2020-04-07:
dennisscholing:

thanks thanks you very much

OndraCerv, 2020-04-12:
Hi, 

i tested with you solution and it doesn't work for me.
First layer calibration is excelent and first layer print i need still  manual offset z axis +- 250-400 um.

I'll try to go back to fw 4.0.3.

mlgold, 2020-04-13:
> > Now running on 4.0.5
> > (the Gcode puts the head in the middle of the bed during warming up and then waits 80sec, before leveling the bed)
> > if you have already loaded filament you have to wait longer 120sec + / _)).
> Modified start G-code:
> 
> ```

This works very well, first layer is very consistent now. Even with 4.04 it was not that great. I don't see how this does any harm, besides a couple more minutes wait, so I expect to keep these changes even with the next firmware (unless the next firmware does the same thing without needing it in gcode)

dahuafschmied, 2020-04-14:
But be careful. You do movement g-codes before homing!

dennisscholing, 2020-04-14:
> But be careful. You do movement g-codes before homing!

Please read the G-code carefully, it is homing first :)
`G28 ; home all`

dahuafschmied, 2020-04-14:
> > But be careful. You do movement g-codes before homing!
> 
> Please read the G-code carefully, it is homing first :)
> `G28 ; home all`

ups. my fault

OndraCerv, 2020-04-15:
HI, 

how to make fw down to 4.0.3. Printer does not offer firmware from USB flash disk after reboot nor from the menu.

Thanks

dinands, 2020-04-15:
when you turn on the mini press the dialknob it asks if you want to flash

feutl, 2020-04-24:
I wanted to jump onto this as well, but only because I am sing 4.0.5 and do not have the problem.
All my prints stick to the bed properly using PLA and ASA. But flipping the sheet to the other site makes the first layer unusable, which also makes no sense to me.

Perhaps this is somehow related? I am using a smooth steel sheet.

Texy, 2020-05-13:
The problem is that the MINDA does not have temperature compensation like the PINDA of the MK3/s. The MINI motherboard supports a thermistor feed from the MINDA, but it is not present. Would be good if PRUSA supplied an upgraded MINDA with thermistor, or allowed the PINDA to be fitted with an upgrade kit. It's pretty painful to get the Z level consistently correct without it. I have no such problems with my MK3.

Dewolla, 2020-05-18:
I have my mini in an enclosure in my workshop and I always pre-heat the enclosure before printing. getting the ambient temp to around 66-68 degrees F. prior to starting the print allows for the MINDA to come up to temp. I also like others here added a home all g-code so the sensor is closer to the print bed while it's starting up. (so for me it homes all then does the heat up). 

I didn't realize the MINDA didn't have temp compensation, and good to know the PINDA sensors do. 
I agree would really like to have a temp compensation feature available as an upgrade.

BlueDragonX, 2020-07-07:
I've encountered this problem as well, but I'm unable to fix it through the described preheating method due to my Mini being in an enclosure. Instead I have to ensure that the enclosure is heated to 50*C before home and bed leveling. This is particularly tricky since I'm using the bed heater to warm the enclosure.

vazmuten, 2020-09-11:
I also experience the same MINDA problems and have to follow a slow preheating vodoo magic procedure in order to properly start any printing process. Also if the printing process is PAUSED and the temperature goes down, then after the RESUME MINDA causes a complete disaster by erroneously measuring the height. I also insist on urgent firmware fix or better an upgrade kit with any reliable height sensor.

JonnyBooker, 2020-09-25:
Is there any further update on this issue? I have noticed the exact same problem. The pre-emptive preheating is the way I'm having to use to solve this issue but it is a pain to do everytime, especially if something goes wrong, you can't get restarted again very fast

JohnnyDeer, 2020-10-05:
We are constantly working on solution, unfortunately it´s long term issue because is touching hardware. Changes and functions of FW did not solve this completely. Thank you for understanding.

mgcrea, 2020-10-09:
I'm encountering this as well on a brand new mini (shipped end of Aug). I think the priority should be to document upgrade to alternative more precise hardware (PINDA/BLTouch), It's a pain to have a random Z-height at every print start even if the drift is small, you often end up with failed print due to improper adhesion that often is only visible later in the print.

DRKMSTR, 2020-10-25:
> I'm encountering this as well on a brand new mini (shipped end of Aug). I think the priority should be to document upgrade to alternative more precise hardware (PINDA/BLTouch), It's a pain to have a [...]

This is critical for my application, I am trying to run a small business around printing and I am trying to transition over from Ender 3 Pro's to Prusa Mini's, however the time necessary to start a print on a Prusa Mini (I have to monitor and adjust first layer offset, EVERY SINGLE PRINT) is keeping me from upgrading.  I can set and forget an Ender 3 for 50+ prints with no problems whatsoever.

If they would add optional support for the PINDA v2 probe (with temperature sensor) I would buy a couple more Mini's immediately.

giordanososa, 2020-10-29:
There is clearly an issue here but it is hard to say it is only due to the lack of thermistor in the probe. 
On one hand, using the warm-up script provided by @dennisscholing (in my case) completely solved the issue. I have had no problems since doing that change.

On the other hand, I had had no problems the first 2 months of use, never had to tweak my first layer. But that changed a few weeks back, all of a sudden I had to monitor and teak every single first layer print, or just wait for it to fail and start again, which constantly proved successful in the 2nd attempt. My impression is that the change matches the upgrade to firmware 4.2.1 but that part is anecdotal.

In any case, it just being a lack of thermistor does not explain how me (and others) did not see the issue for quite some time, but having clearly different results when warming up the probe strongly suggests something must be done about the temp.

Here are my results with/without warming up (in both cases, first layer calibration turned out OK and with the same value).
In the image, you can see first layer calibration results using original print, on the right the same print with start-up script to allow the probe to warm-up, and farther left the delta between both ABL, up to 0.121mm difference between both in some points.

[image]

DRKMSTR, 2020-10-29:
@giordanososa I've used a startup script (well, more or less a startup procedure + some custom GCODE) and it helps when I only print one type of material over a long period....however my issue is when printing multiple types of materials. I often switch between PLA: 210C/60C, ASA: 260C/100C, and PETG: 230C/85C and it takes 3-5 prints for it to normalize, but by then I'm switching materials again.

So for my use-case, it's extremely problematic. I'm trying to start up a business and the inconsistencies are really eating into my time. I'm this close to switching to Ender 3 Pro's (MW Power Supplies) and tossing a BL-touch probes on them, it's just slightly more time-intensive right now. Heck, I'll do that this weekend and see if it's a decent temporary fix. I'd prefer to buy more PRUSA mini's though.

vidratajec, 2020-11-22:
Can we use PINDA v2 on the mini?

DRKMSTR, 2020-11-23:
> Can we use PINDA v2 on the mini?

No, we can't.  Welcome to the struggle.

json81, 2020-11-24:
I have the same problem. Heating up the MINDA works much better for me as well.

Prusa has announced a solution today. **SuperPINDA** is the replacement and is now available as an upgrade for Prusa Mini and MK3S.

https://shop.prusa3d.com/en/original-prusa-mini/1399-original-prusa-mini-to-mini-upgrade-kit.html

JohnnyDeer, 2020-11-25:
Solution is here, thank you for patience. As you may noticed we released MINI+ with a new Mesh Bed Leveling sensor, the SuperPINDA. SuperPINDA will make calibration of the first layer much more consistent. So all of you with inconsistent first layer issue please contact our Support and we will replace your MINDA for SuperPINDA. I will close this issue soon along with all related. In any new cases of this bug please create new issue to make this repositary clear. Thank you all.

JohnnyDeer, 2020-11-25:
Solution is here, thank you for patience. As you may have noticed we released MINI+ with a new Mesh Bed Leveling sensor, the SuperPINDA. SuperPINDA will make calibration of the first layer much more consistent. You can find it in the e-shop as an upgrade for your MINI to MINI+: https://shop.prusa3d.com/en/original-prusa-mini/1399-original-prusa-mini-to-mini-upgrade-kit.html
If anyone faces issues with inconsistent first layer and you have explored all options first and suspect MINDA to be the cause, please contact our Support with sufficient proof, photos or a video that your issues could be MINDA . I will close this issue soon along with all related. In any new cases of this bug please create new issue to make this repositary clear. Thank you all.

JohnnyDeer, 2020-12-07:
I close this issue, all new cases with SuperPinda please add to #1131 Thank you.

closed 2020-12-07 by JohnnyDeer

vladionescu, 2022-03-30:
Unfortunately I was told by support that while this is a known issue, I'm on my own to buy the SuperPINDA ($26) and shipping ($15 cheapest DHL). This isn't covered under warranty, and there are no discounts or help fixing this long standing flaw.

Extremely disappointing to say the least. I bought a Prusa because I wanted a printer I didn't have to babysit for the first layer, and because I expected good support should something go wrong. My expectations were misguided :) alas.

apos, 2023-09-01:
> @vladionescu : Did you had success with the super pinda? Is the problem with different layer heights for different temperatures really gone?